sakaki-/gentoo-on-rpi-64bit #1: Unable to log in with passwd raspberrypi64

NeddySeagoon, 2017-01-30:
Step 1. I wrote the image to a 32GSD card, made a 2G swap at the end of the card then changed the console and Xorg keymaps to dvorak-uk. Just to save finding a qwerty keyboard.
 
Step 2. Moved the SD card to the the Pi and it worked as advertised, until the screen locked.
I also tried su - and sudo su - (before screen lock) both of which gave authentication failures. 
The default user is in the wheel group but not in the sudoers file.

The workaround is to replace the password hashes in /etc/shadow on a PC.

sakaki-, 2017-01-30:
Strange - I could replicate this (i.e. could not `su - root` when logged in to X as `demouser`) but _was_ able to ctrl-alt-F1 and log in as root on a text console on the Pi, and also over `ssh`.
Also, xscreenlock should be disabled for `demouser`?
I will have a look at the password hashes in case something is up with them. Will put `demouser` in the `sudoers` file also for a re-release.

NeddySeagoon, 2017-01-30:
I was definitely demouser, I could not get to root and I have not yet created a new user.
I was in the middle of doing something, so I suspect it was the time of day issues associated with swclock that triggered the screen lock.
The Pi has been idle for about 3 or 4 hours, with the demouser auto logged in. Its not locked again.
Maybe it was finger trouble my end?

sakaki-, 2017-01-30:
It is possible there is also some odd interaction caused by large initial clock reset caused by chrony. Will put the timestamps in `/etc/shadow` back to some very safe value in 1.0.1; hopefully will obviate this.

sakaki-, 2017-01-30:
It is possible that the large time-step caused by chrony is also triggering something. Will see what happens on this next release.

sakaki-, 2017-02-06:
Note that if you have been logged in for a while (few days) and you are bounced to the the `lightdm` login screen (as opposed to `xscreenlock`), and seem unable to login from there (even though your password is correct), it is likely that the CMA has run out of memory (there appears to be a small leak in there somewhere in vc4). If this is the case, you can see in `dmesg` (do ctrl-alt-F1 to get a text console, and login as root to check):
```
[78730.286091] vc4-drm soc:gpu: failed to allocate buffer with size 4227072
[78730.300899] vc4-drm soc:gpu: failed to allocate buffer with size 4227072
[78730.300959] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[78730.300964] [drm] num bos allocated: 38
[78730.300968] [drm] size bos allocated: 21648kb
[78730.300972] [drm] num bos used: 37
[78730.300977] [drm] size bos used: 17520kb
[78730.300981] [drm] num bos cached: 1
[78730.300985] [drm] size bos cached: 4128kb
```
This seems to require a reboot to fix atm. There are reports of similar issues on vc4 (google "vc4-drm soc:gpu: failed to allocate buffer") so hopefully will be fixed shortly. Will keep an eye on it.

hamdyaea, 2017-02-10:
I have the same problem on my Raspberry Pi 3. The password raspberrypi64 don't work. I can be root with demo user by using #sudo su
But't impossible to change the password for demouser or root.

sakaki-, 2017-02-10:
If you do `sudo su --login` can you then use `passwd` to set the root password? If not, what error is shown?
Also, can you do `<ctrl><alt><F1>` and login at the text console?

3PO, 2017-02-25:
>I have the same problem on my Raspberry Pi 3. The password raspberrypi64 don't work.

Same Problem here. :(

sakaki-, 2017-02-25:
@3PO could you do `sudo su --login`, or `<ctrl><alt><F1>` and login at the text console as root?

3PO, 2017-02-25:
>@3PO could you do sudo su --login, or <ctrl><alt><F1> and login at the text console as root?

No.

But i solved it by editing the shadow file. ;)

sakaki-, 2017-02-25:
Hmm. Well glad you got it working anyway. This is a bit of an odd issue - it is affecting only some users but is definitely happening. Will look at this again over the next few days. Maybe creating the image's `/etc/shadow` entries from a linux PC as part of the image packing script would be more reliable.

3PO, 2017-02-25:
You can do the following:

Put the SD-Card into a Linux PC and mount the root partion.
Edit the "/etc/shadow"

Replace

    root:$6$1hDzaXrB$SRhr2jcZIRCcrIGvg9NLfqTbsY9Qh9TMiBqiOYQvHspNb9nEdZDwliyxDPEKRy7w4j1hzRFEmR5zQ1F9Oe.qS1:17172:0:::::

with:

    root::17172:0:::::

--> delete all between the first and the second colon.

After that, you can login as root, without a password.

Then you can set a new password with:

    passwd

sakaki-, 2017-03-04:
Released [v1.0.2](https://github.com/sakaki-/gentoo-on-rpi3-64bit/releases/tag/v1.0.2): this has a [modified](https://github.com/sakaki-/gentoo-on-rpi3-64bit/commit/29ad2fd6bcf2cf62d575b5e823a1cc12793e12d9) `autoexpand_root_partition.start`, which force-sets `root` and `demouser` passwords on first boot. Hopefully will finally close out this issue.

3PO, 2017-03-10:
This does not work!

I cannot login a root.

Why do you not simply distribute the image without a root password?

--> See the HowTo i wrote above.

sakaki-, 2017-03-10:
Yes, I saw your howto, thanks for posting. However, Gentoo appears to set a default root password when the first autologin happens in this manner, so it doesn't really help.
You will see that my `autoexpand_root_partition.start` calls:
```sh
DEFAULTPW="raspberrypi64"
...
chpasswd <<<"root:${DEFAULTPW}"
chpasswd <<<"demouser:${DEFAULTPW}"
```
That's pretty straightforward, and it works on several systems I have tested it on. If that file is executed and you _still_ cannot login to root, then something else must be up. Keyboard mapping maybe?

3PO, 2017-03-10:
> ... Keyboard mapping maybe?
No!
 
Your "auto set password script" don't work.

So i cannot your your distrie.

3PO, 2017-03-10:
> However, Gentoo appears to set a default root password when the first autologin happens in this manner, so it doesn't really help.
You will see that my autoexpand_root_partition.start calls: ...

I wrote little Patch for fixing this problem:

        --- autoexpand_root_partition.start.orig	2017-03-10 21:03:13.905241416 +0100
    +++ autoexpand_root_partition.start	2017-03-10 21:49:09.000000000 +0100
    @@ -32,7 +32,6 @@
     ROOTPART="${ROOTDRIVE}p2"
     SENTINEL1="/boot/autoexpand_root_partition"
     SENTINEL2="/boot/autoexpand_root_filesystem"
    -DEFAULTPW="raspberrypi64"
     
     if [ -f "${SENTINEL1}" ]; then
     	# first boot, need to resize /dev/mmcblk0p2 (root) to fill disk
    @@ -65,9 +64,9 @@
     		exit 1
     	fi
     	# now, change passwords
    -	cecho "Changing root and demouser passwords, to '${DEFAULTPW}'"
    -	chpasswd <<<"root:${DEFAULTPW}"
    -	chpasswd <<<"demouser:${DEFAULTPW}"
    +	sed -i /etc/shadow -e "s/^\(root:\)[^:]*:/\1:/"
    +	sed -i /etc/shadow -e "s/^\(demouser:\)[^:]*:/\1:/"
    +	sed -i /etc/ssh/sshd_config -e "s/#PermitEmptyPasswords.*/PermitEmptyPasswords yes/"
     	cecho ""
     	cecho "*******************************************************************"
     	cecho "*  Resize completed - rebooting now (this will only happen once)  *"

Now i can login without a password as user root and as user demouser local an as well via ssh.

sakaki-, 2017-03-10:
OK I get that, but for interest, why do you think `chpasswd <<<"root:foo"` does not set the root password to `foo`?

3PO, 2017-03-10:
I don't know, because i didn't further investigate it.  

If i set `"DEFAULTPW="foo"` i cannot login as root with password "foo".

My solution is working for me and i can set my own password, if i want. :)

sakaki-, 2017-03-11:
If you do (as root, once you have successfully logged in) `chpasswd <<<"root:foo"`, does that change the password for you? I'd be keen to understand if its something to do with the initial autologin since it doesn't appear to happen on all systems.

sakaki-, 2017-07-15:
Closing - the password change is now [forced](https://github.com/sakaki-/rpi3-overlay/blob/master/sys-apps/rpi3-init-scripts/files/init.d_autoexpand_root-2#L75-L78) for `root` and `demouser` by the OpenRC `autoexpand-root` startup service, on version 1.1.0 of the image (even if [opting out](https://github.com/sakaki-/rpi3-overlay/blob/master/sys-apps/rpi3-init-scripts/files/init.d_autoexpand_root-2#L128-L138) of the partition resize).
Please re-open if you still have problems

closed 2017-07-15 by sakaki-