sakaki-/gentoo-on-rpi-64bit #2: CPU Frequency Scaling

NeddySeagoon, 2017-01-30:
The default cpufreq/scaling_governor is powersave, so as distributed, the CPU clock is always 600MHz.
The "on demand" works but it needs to be selected as default in the kernel or set as a part of the boot.
This switches the CPU between 600MHz and 1.2GHz.

sakaki-, 2017-01-30:
I'm ttrying to stay with default kernel config if at all possible...

Would the `etc/init.d/raspi-config` the settings from Raspbian be appropriate ([link](http://raspberrypi.stackexchange.com/questions/9034/how-to-change-the-default-governor))? If so I can add a startup service to the image to the image to set something like:
```sh
SYS_CPUFREQ_GOVERNOR=/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
if [ -e $SYS_CPUFREQ_GOVERNOR ]; then
    echo "ondemand" > $SYS_CPUFREQ_GOVERNOR
    echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
    echo 100000 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_rate
    echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_down_factor
    echo 1 | sudo tee /sys/devices/system/cpu/cpufreq/ondemand/io_is_busy
fi
```

NeddySeagoon, 2017-01-30:
I use ondemand with the default settings but its set in my kernel.
Staying with the default kernel config is a good idea. It makes for ease of maintenance

sakaki-, 2017-01-31:
Closing per your forum post [here](https://forums.gentoo.org/viewtopic-p-8024608.html#8024608).

closed 2017-01-31 by sakaki-

leio, 2017-02-04:
You could provide your own configuration overrides, dropped into kernel/configs of the kernel source tree, and then pass that to the make defconfig_bcm* call, e.g a
kernel/configs/sakaki.config file with the content

```
# Use ondemand by default
CONFIG_CPU_FREQ_DEFAULT_GOV_ONDEMAND=y
```

and then call

`make bcmrpi3_defconfig sakaki.config`

and it should apply the override to it.

sakaki-, 2017-02-05:
@leio - agreed. The reason not to I guess is that 'official' prebuilt Raspbian kernels / module sets may in time become available for `arm64` (as for `arm`) (using `bcmrpi3_defconfig`) and it'd be nice if we could have the option to use them. Raspbian itself sets the governor to `ondemand` during the boot process; on the current Raspbian image, `/etc/init.d/raspi-config` reads:
```sh
#!/bin/sh
### BEGIN INIT INFO
# Provides:          raspi-config
# Required-Start: udev mountkernfs $remote_fs
# Required-Stop:
# Default-Start: S
# Default-Stop:
# Short-Description: Switch to ondemand cpu governor (unless shift key is pressed)
# Description:
### END INIT INFO

. /lib/lsb/init-functions

case "$1" in
  start)
    log_daemon_msg "Checking if shift key is held down"
    if [ -x /usr/sbin/thd ] && timeout 1 thd --dump /dev/input/event* | grep -q "LEFTSHIFT\|RIGHTSHIFT"; then
      printf " Yes. Not enabling ondemand scaling governor"
      log_end_msg 0
    else
      printf " No. Switching to ondemand scaling governor"
      SYS_CPUFREQ_GOVERNOR=/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
      if [ -e $SYS_CPUFREQ_GOVERNOR ]; then
        echo "ondemand" > $SYS_CPUFREQ_GOVERNOR
        echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
        echo 100000 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_rate
        echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_down_factor
      fi
      log_end_msg 0
    fi
    ;;
  stop)
    ;;
  restart)
    ;;
  force-reload)
    ;;
  *)
    echo "Usage: $0 start" >&2
    exit 3
    ;;
esac
```

That's where I got the settings used in the '.start' script from. Arguably we should have a similar OpenRC/systemd service available on Gentoo.

leio, 2017-02-06:
Well, you can still use them then, the config override just switched to ONDEMAND, if the arm64 upstream config will already be ondemand, it'll just be a no-op. Arguably it's better to set the default in kernel, so that the boot can go faster before too, not only after the init script gets ran.