sakaki-/gentoo-on-rpi-64bit #3: Consider passing -mcpu in CFLAGS

leio, 2017-02-04:
I don't know if there's any practical difference, with cortex-a53 being on the low end of arm64 world, but might want to consider passing -mcpu=cortex-a53 instead of -mtune=cortex-a53 (or in addition, but the -mcpu should imply the -mtune). This will ensure any instructions that only the A53 has will be used when beneficial, instead of just tuning for the CPU family instruction latencies and the like.

And while on the subject of make.conf, to not be in potential trouble, might want to at least USE=bindist.

sakaki-, 2017-02-05:
Hi - is that correct? The [gcc manual](https://gcc.gnu.org/onlinedocs/gcc/ARM-Options.html) seems to imply that `mcpu` is just a way of setting `march` and `mtune`; after all:
> Permissible names for this option (-mcpu) are the same as those for -mtune.

and:

> GCC uses this name (i.e. that of -mcpu) to derive the name of the target ARM architecture (as if specified by -march) and the ARM processor type for which to tune for performance (as if specified by -mtune). Where this option (i.e., -mcpu) is used in conjunction with -march or -mtune, those options take precedence over the appropriate part of this option.

Also from this [stackoverflow answer](http://stackoverflow.com/questions/29428726/selecting-appropriate-arm-mfpu-option-in-gcc-based-on-cpu-features/29432456#29432456):
> -march and -mtune (or -mcpu as a shorthand for both) [...]

However I may well be wrong on this point; if you have a reference that points to `mcpu` changing code generation on arm64 (wrt the equivalent `march`/`mtune`) I would be very interested to see it.

On the `make.conf` point, the version used when building the image may be seen [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit/blob/master/reference/make.conf). As you can see, bindist is indeed set (that's why the image has the 'Aurora' Firefox, for example); the USE flags (in addition to those from the `arm64` desktop profile are:
```sh
USE="bindist -mudflap -sanitize"
USE="${USE} bluetooth egl gles1 gles2 lock thunar qt4 ffmpeg"
USE="${USE} -gnome -kde"
```

leio, 2017-02-05:
Ok, the documentation there says things a bit different (or clearer) than the man page. So it might be that if you set -mcpu, you don't need to set anything else; -march is inferred and -mtune as well. So if that's how it is, something like -mcpu=cortex-a53+crc would be sufficient and it'll automatically figure out -march=armv8-a+crc from there and -mtune is without question also implied by -mcpu.
Thanks for the documentation pointer; better data point for me to figure this out for sure now (need to confirm -mcpu doesn't do anything other than the -march and -mtune inferring in terms of instruction usages and so on; maybe this is simple like this for now but will get more complicated in the future?)

Regarding bindist, I think I read a wrong make.conf from somewhere or was just blind then.

closed 2017-02-05 by sakaki-