sakaki-/gentoo-on-rpi-64bit #4: Strange "No such file or directory" Problem

3PO, 2017-02-26:
Hi,

I want to watch the CPU temperature. 
I followed this Howto:
http://www.funtoo.org/Raspberry_Pi_Userland_(VCGENCMD)

But this does not work. 
When i call the "vcgencmd", i get this issue:

    pi64 ~ # /opt/vc/bin/vcgencmd measure_temp
    -bash: /opt/vc/bin/vcgencmd: No such file or directory
    pi64 ~ #

But the File exists:

    pi64 ~ # ls -lah /opt/vc/bin/vcgencmd
    -rwxr-xr-x 1 root root 15K Feb 26 16:32 /opt/vc/bin/vcgencmd
    pi64 ~ #

Where is my fault?

sakaki-, 2017-02-27:
The `vcgencmd` on the image is non-operational; none of the standard versions in the `arm64` tree currently work. It _can_ be made to run, but you need the custom `media-libs/raspberrypi-userland-9999` ebuild from my `rpi3` repo. To get this, issue:
```console
pi64 ~ # emaint sync --repo=rpi3
pi64 ~ # emerge -av =raspberrypi-userland-9999
```
You will need to unmask a few packages `**` to get this to build, as prompted by `emerge`.
Once installed, you should be able to run `vcgencmd` OK.
hth, sakaki

3PO, 2017-03-01:
Works fine. :) Thanks!

    pi64 ~ # vcgencmd measure_temp
    temp=51.5'C
    pi64 ~ #

I installed it, by doing the following:

    emaint sync --repo=rpi3
    echo "=media-libs/raspberrypi-userland-9999 **" >> /etc/portage/package.keywords/raspberrypi-userland
    emerge -av =raspberrypi-userland-9999

closed 2017-03-01 by 3PO