sakaki-/gentoo-on-rpi-64bit #8: Supporting ARM32 programs

Myriachan, 2017-04-25:
Is it possible to set up the installation to have a 32-bit library set in /lib so that ARM32 programs can run?  I'd like to test both ARM32 and ARM64 code using my Raspberry Pi 3.

I suppose that recompiling the kernel to enable the ARM32 compatibility mode would also be required.

necrose99, 2017-05-23:
dual ABI , would be a nicety... ,as say @Sabayon has an automated package builder ....

necrose99, 2017-06-05:
[linux-4.11.3-gentoo.kernel-arm64-dual-abi.config.txt](https://github.com/sakaki-/gentoo-on-rpi3-64bit/files/1053411/linux-4.11.3-gentoo.kernel-arm64-dual-abi.config.txt)

this is after emergeing gentoo kernel , and compiling it.  over ssh... 
anycase its not Fast... 
ssh send commands get on ps4 or MMO of choice check back in a matter of hours... 
this is a generic ARM64 kernel includes RPI3 , arm64-servers/desktop dev boards...., etc....

sakaki-, 2018-04-17:
I've just posted a wiki entry about how to do this, [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit/wiki/Running-32-bit-Raspbian-Packages-on-your-64-bit-Gentoo-System).

Screenshot of 32-bit Raspbian Lazarus running on an 64-bit Gentoo RPi3 B+:

[image]

sakaki-, 2018-04-23:
Hopefully everyone who needs this has seen it now (or will find it on the wiki).
Closing.

closed 2018-04-23 by sakaki-