sakaki-/gentoo-on-rpi-64bit #9: Unable to boot a different kernel

Yamakuzure, 2017-05-04:
Hi there,

first of all: Great work! The image works like a charm, and my RPi3 was rock stable, no matter what.

However, I wanted to upgrade to the current 4.10.y sources, and did so like in the README.

(_Note_ : sys-kernel/raspberrypi-sources-4.10.9999 is in the tree, so it does not need to be pulled manually.)

I did substitute kernel8.img and the corresponding dtb[o] files.
But on next boot I was greeted with the multi colour screen of death.

So I reverted back to the old kernel8.img (I always keep backups of the kernel and all dtb[o] files) and powered the PI back on. Willing to find out what went wrong. I basically used the recommended defconfig, so I was a bit puzzled.

However, it did not come back on.

Then I remembered, that I deleted two other images in /boot.
Could it be that the thing is configured (somehow, because kernel=kernel8.img in config.txt doesn't help) to use an alternative kernel name, and doesn't accept different ones?

Oh, and how do I mount the genpi64.img? I wanted to copy the deleted kernel images back, but it seems to be too early with no decent coffee here, and I am completely at sea.

Thanks in advance

Sven

Yamakuzure, 2017-05-04:
okay, mind it.

I totally forgot, that the first thing I did was to add dynamic cma to cmdline.txt and config.txt. _And_ it seems that the 64bit mode is quite allergic to that. ;-)

closed 2017-05-04 by Yamakuzure

necrose99, 2017-06-29:
@Yamakuzure I was able to build the Gentoo 4.x Kernel on RPI3 , including the RPI-3 firmaware 

unfortunately he hasn't a  /boot / and /efi     (/boot/efi)  as you could symlink kernel @ .bin and add to boot. for testing.  but I have a docker image built..