sakaki-/gentoo-on-rpi-64bit #10: Minimal install

kde99, 2017-05-06:
Can I install this without a GUI?

sakaki-, 2017-05-10:
Simplest thing for now is probably just to install the default system, turn off the GUI, and then (optionally) uninstall the GUI elements you don't want.

To turn off the GUI, first boot normally, then log in as `root` to a virtual text console (using e.g. <kbd>ctrl</kbd><kbd>alt</kbd><kbd>F1</kbd>), then issue:
```console
# /etc/init.d/xdm stop
# rc-update del xdm default
```
Once you do this, by default you will be landed at a standard text console on subsequent boots (or you can of course use `ssh` to connect).

Having done this, you can then (optionally) delete any of the installed packages you don't want, by using
```console
# emerge --depclean --ask --verbose <pkgname1> <pkgname2> ...
```
so, for example
```
# emerge --depclean --ask --verbose www-client/firefox mail-client/claws-mail
```
You can see a full list of (top-level) packages [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit/blob/master/reference/world-packages) (or just look in the file `/var/lib/portage/world`).
Note that because `emerge --depclean` doesn't do anything if the package you cite is a dependency of another installed package, you will probably need to e.g. cite all the `xfce` packages in a single `emerge --depclean` invocation.

Once the packages have been removed, you can then clean up your system with:
```
# emerge --depclean          # NB no arguments this time; removes unreferenced packages
# revdep-rebuild             # fix any shared-library dependencies the above may have broken
# emerge @preserved-rebuild  # fix packages depending on stale libraries
# env-update                 # update environment settings
```

hth, sakaki

kde99, 2017-05-10:
@sakaki- Thank you so much!! Would it work if I do this in chroot then write it to my pi?

sakaki-, 2017-05-11:
You would need to `chroot` into the mounted image from an RPi3 running a 64-bit kernel (since the userland programs are ARM-specific binaries (which are not x86_64 compatible), and compiled in 64-bit mode), but if you do that it should work. The default Raspbian system has a 32-bit kernel, so you can't use that as a host OS unfortunately.

Simplest however would be to boot the system and proceed as above (running natively on the Pi3). All the steps up to the `revdep-rebuild` can be done _without_ network connectivity (if you are concerned about the security of the GUI elements).

sakaki-, 2018-04-23:
I have now written up a [short wiki article](https://github.com/sakaki-/gentoo-on-rpi3-64bit/wiki/Create-a-Slimmed-Down-Version-of-the-Image) about the "slim down" process (and [added some notes](https://github.com/sakaki-/gentoo-on-rpi3-64bit#rpi3_headless) into the main readme too, regarding headless running).
Closing.

closed 2018-04-23 by sakaki-