sakaki-/gentoo-on-rpi-64bit #12: extra goodies

necrose99, 2017-05-23:
https://plus.google.com/communities/111215969224831918857
https://www.dropbox.com/sh/4u9gbhps0ek402e/AAC5VLBE32cwH2gGjkj0Yz9sa?dl=0

note some the sabayon packages work-ish but kinda not...  equo/entropy/molecule..... 
Zenmap gui dose work... 
golang , etc... 
hopefully can get docker packages to build..... so you can cluster more easily with other pi's... 

Jenkins .... might be nice to test.....  , for obvious reasons , can automate a few emerges..... 
slowly adding a few pkgs form [@pentoo](https://github.com/pentoo/pentoo-overlay)

necrose99, 2017-05-23:
anyhow I've uploaded to dropbox , can be tested. ...  (I need to get me a server .....) but for now 
enjoy ... packages we built on the pi3....

necrose99, 2018-07-14:
[Packages.txt](https://github.com/sakaki-/gentoo-on-rpi3-64bit/files/2194511/Packages.txt)

https://www.dropbox.com/s/1r3dnawxc46lzq6/packages_7-8-2018.tar.xz?dl=0  

few days out of sync but much of KDE however might need a pine/rock64 with 4-GIGs of ram .

viking76, 2018-08-08:
i have worked to compile nodejs with no success on -arm64, who have a success with this ebuild ?

necrose99, 2018-08-08:
https://www.dropbox.com/preview/sakaki--gentoo-on-rpi3-64bit/packages-rpi3-arm64/net-libs/nodejs/nodejs-8.11.2-1.xpak?role=personal

viking76, 2018-08-08:
no access

necrose99, 2018-08-08:
https://discord.gg/5yhY9bg  jump into pentoo Linux room.

necrose99, 2018-08-08:
[nodejs-8.11.2.tbz2.zip](https://github.com/sakaki-/gentoo-on-rpi3-64bit/files/2271956/nodejs-8.11.2.tbz2.zip)

sakaki-, 2019-08-17:
Old issue, closing now in pre 1.5.0 housecleaning

closed 2019-08-17 by sakaki-