sakaki-/gentoo-on-rpi-64bit #13: Converting to NOOBS format

procount, 2017-06-27:
I am trying to convert your image to be installable by NOOBS/PINN.
If I remove the `/boot/autoexpand_root_partition` sentinel file as recommended then the raspberrypi64 password does not get set.  In this case what is the default password for root to login?

As a workaround, I was modifying the `etc/local.d/autoexpand_root_partition.start`so that the password and graphical login would be set without partition resizing. But for some reason my root partition always ends up being r/o so that the script will not be effective. I'm still investigating that, but any ideas why this partition should end up readonly would be welcome.

sakaki-, 2017-06-27:
The default password is _supposed_ to be `raspberrypi64` anyway, but this was causing issues for some people (see #1) so I added the belt-and-braces setting to the `autoexpand_root_partition.start` file. In any event, you can always do `sudo su --login root` as demouser to get a root prompt (given the `/etc/sudoers` setup on the image), then set your own password.
No idea why your / should be ro though. Is your root partition `ext4`, as set up in `/boot/cmdline.txt`?

sakaki-, 2017-06-27:
One thought - on Gentoo under OpenRC, the file `/etc/init.d/root` is what brings the root filesystem to a writeable state (it is ro in very early boot). This init script may not be be being executed (although in this case you'd expect the system to fail to come up), or your / may have failed an fsck and been retained as ro for that reason... have you tried checking the / filesystem (on a PC, when not mounted)?

sakaki-, 2017-06-27:
I have merged your PR, thanks, it'll be on the next release (which I am working on at the moment).
When you use your NOOBS fork to boot, and have the ro filesystem, are there any hints about what went wrong in `dmesg`?

procount, 2017-06-27:
I finally managed to log in again, without it immediately requesting me to reset my password.
As gentoo boots it gets to the 
` * Checking local filesystems ...` part
The boot partition checks out fine.
But then it checks the rootfs...
`* Remounting root filesystem read/write ...`
`MOUNT: / not mounted or bad option`
`...`
`Root filesystem could not be mounted read/write`

The fstab looks fine to me. I only replaced mmcblk0p2 with mmcblk0p7 for root, and similarly for boot.
```# NOTE: If your BOOT partition is ReiserFS, add the notail option to opts.
/dev/mmcblk0p6          /boot   vfat    defaults                0       2
/dev/mmcblk0p7          /       ext4    defaults,noatime        0       1
/var/cache/swap/swap1   none  swap sw    0   0
```

The only thing I spotted in dmesg was:
`[  8.966408] EXT4-fs (mmcblk0p7): Unrecognized mount option "freely." or missing value.
I couldn't find any other logs (because /var is read-only?)

sakaki-, 2017-06-27:
`/etc/fstab` looks OK, but that dmesg is odd. What is in `/boot/cmdline.txt` after your script has modified it?

procount, 2017-06-27:
`dwc_otg.lpm_enable=0 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait`

sakaki-, 2017-06-27:
Looks OK but something may be at issue elsewhere - what does `cat /proc/cmdline` show once you are booted (on the ro root system)?

procount, 2017-06-27:
`8250.nr_uarts=0 cma=256M@256M dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1680 bcm2708_fb.fbheight=1050 bcm2709.boardrev=0xa02082 bcm2709.serial=0x7882aa2b smsc95xx.macaddr=B8:27:EB:82:AA:2B bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
`

Maybe it's the way I created the rootfs partition? 
I used `mkfs.ext4` with option `-O ^huge_file` and labeled the partition `root_gen`

sakaki-, 2017-06-27:
Looks OK. Must be a filesystem issue then, since the kernel is unchanged.

Possibly you have run out of inodes? Gentoo requires more than most distros because of the from-source nature. Try running `df -i` (when booted to the ro system) and see what you get for `IUse%` on `/`. If full then you probably have a corrupt root filesystem. It can happen with EXT<n> systems on small (e.g. sd-card) partitions under Gentoo (not an issue on large hard drives with the standard `mkfs.ext4` settings).
As suggested [here](https://wiki.gentoo.org/wiki/Raspberry_Pi_3_64_bit_Install#Make_Filesystems), you could try using
```console
mkfs.ext4 -i 8192 -t ext4 /dev/mmcblk0p7
```
to get more inodes. Also try leaving off the huge_file option, at least initially. The partition label should not matter.

procount, 2017-06-27:
/ has IUse%=22%
Inodes 1937712
IUsed 414794
Ifree 1522918

sakaki-, 2017-06-27:
OK, not that then, but must be _something_ to do with the root filesystem - the kernel is the booting OK and I assume there is nothing weird going on in config.txt, and the full (`/proc/cmdline`) kernel command line matches a normal startup. The `dmesg` output
```
[ 8.966408] EXT4-fs (mmcblk0p7): Unrecognized mount option "freely." or missing value.
``` 
is definitely indicating something is wrong with root. Perhaps if you try making `/dev/mmcblk0p7` without huge_file?

One last thing you could try is, when the ro system comes up and you are logged in, to try to manually remount the `/` filesystem read-write - you might get a better diagnostic from that as to what it thinks is wrong:
```console
pi64 ~ # mount -v -o remount,rw /
```

Sorry this is chewing up so much of your time! If you still can't get it to start after this then if you can send me an image of your boot and root filesystems (`/dev/mmcblk0p6` and `/dev/mmcblk0p7`), I'd be happy to look at it this end for you.

PS I assume you have rebuilt the container image since the original-sentinel version of `autoexpand_root_partition.start`
ran? The beginning of that file contains:
```bash
ROOTDRIVE="/dev/mmcblk0"
ROOTPART="${ROOTDRIVE}p2"
```
and then it attempts to (given the original sentinel setup):
```bash
sfdisk --no-reread -N 2 <<<", +" "${ROOTDRIVE}"
```
and then
```bash
resize2fs -f "${ROOTPART}"
```
either of which is going to have a potentially nasty effect on `/dev/mmcblk0p2`, whatever that happens to be... (I appreciate your PR'd version doesn't do this, but the damage may already be done to the partition table, not sure whether you rebuild the full partition structure each test.)

procount, 2017-06-27:
I already tried without Huge file - no difference. I also tried mounting root manually (with -n instead of -v) but with the same error. I will try again with -v. Actually I'm sorry this is taking up so much of YOUR time. This is normally a straightforward process. In fact I have a script to do it all for me for 16 or so different OSes. I must be missing something obvious. Rest assured I started with a clean image with SENTINEL3 setfor PR, so none of the resizing happened. I do have to keep restarting from scratch, but the installation only takes 10 mins or so. It's if I have to change my tar files from the original image that it takes the time. I appreciate the offer of looking at my images. It may come to that, but I have another couple of tricks to try first. I'll let you know how they go. (Why is my return key not working..... ;-).

procount, 2017-06-28:
I tried -v and it still comes up with the same error. I've still got to try -i 8192 plus some other things....

Assuming I do get this to install properly, I could do with a logo for Gentoo (40x40 pixel png file) plus a few images (about 2-4 png files of size ~384x288 pixels) to be shown as a slideshow during installation, illustrating the main features/benefits of Gentoo. :smile:

sakaki-, 2017-06-29:
OK I'll see what I can put together...
Do let me know if there's anything else I can do to help debug this boot issue.

procount, 2017-06-29:
No rush, especially as I am still struggling. I even extracted the rootfs partition from your image and DD'd it directly to /dev/mmcblk0p7 to eliminate any changes to the format / partitioning, and it was STILL r/o.

I will prepare an image for you so you can have a look at it.
All I am essentially trying to do is to move the boot and root partitions to other partition numbers on the SD card. 
Can you think of any other files that might reference /dev/mmcblk0p2?
I am moving them into logical partitions. Would that make a difference? Maybe I will try locating them in different primary partitions next...

procount, 2017-06-29:
Finally some good news - I managed to get PINN to boot gentoo64!
To do this, I installed gentoo64 to primary partitions 2&3. I'm not sure if this is the only reason, but now I have a working image, I will try installing them to logical partitions.

procount, 2017-06-29:
And now it works on partitions 6&7 as well. 
So I think it's something to do with the original image rootfs failing fsck or something. I had to do an fsck on it prior to writing it to the SD card. The rootfs seems to pass an fsck on ubuntu, but then failed it under gentoo (?) But it's looking promising at last. I just need to make the procedure repeatable!

sakaki-, 2017-06-30:
Good news - well done.

sakaki-, 2017-07-04:
Are there examples of the slideshow images for other O/S installs I could take a quick look at to see the kind of thing you're after?
Various formats of the Gentoo logo itself are available [here](https://www.gentoo.org/inside-gentoo/artwork/gentoo-logo.html).

procount, 2017-07-04:
You can have a look here for a list of OSes.
https://sourceforge.net/projects/pinn/files/os/
 In each OS folder, have a look in the slides_vga subfolder. Some are better than others :) Thanks.

sakaki-, 2017-07-15:
Hi - haven't forgotten about this, but have been busy putting the [1.1.0 release out](https://github.com/sakaki-/gentoo-on-rpi3-64bit/releases/tag/v1.1.0), which has a number of significant improvements (inclusion of packages like libreoffice thunderbird & gimp, binary kernel package support, easier updating etc.)

Would you mind trying to see if the new image works for your installer? I have migrated your PR #14 into an OpenRC service on the image (visible [here](https://github.com/sakaki-/rpi3-overlay/blob/master/sys-apps/rpi3-init-scripts/files/init.d_autoexpand_root-2)) but the logic remains the same - you should be able to use `autoexpand_root_none` rather than `autoexpand_root_partition` sentinel files.

Now I have some bandwidth, I'll put some slides together for you over the next few days, with screenshots from the new image. Thanks!

procount, 2017-07-15:
Hey, no worries! I've been busy myself. Will try the new version shortly. Hopefully it will be easier than last time :) thanks.

sakaki-, 2017-07-20:
OK I have posted six first-cut png files [here](https://github.com/sakaki-/resources/tree/master/raspberrypi/pi3/noobs); is that the sort of thing you need? I can make the background white easily if that works better. (PS the benchmarks quoted are from [these results](http://www.roylongbottom.org.uk/Raspberry%20Pi%20Benchmarks.htm#anchorStart)).
I am going to be putting out a 1.1.1 release, hopefully today, addressing some minor issues with 1.1.0 (the overscan cursor alignment issue #17 being the most annoying); so it'd be worth using that version if you could.
Also, I intend switching the `root` (and `boot`) filesystem to PARTUUID naming in the `cmdline.txt` and `/etc/fstab` (I've had a request to do this for people not booting the image from a microSD card); I guess this shouldn't matter too much to you, as you modify these in your installer anyway?

procount, 2017-07-20:
Looking good. Thanks!
And I'd just started converting v1.1.0..... :) OK. I'll hold off til the next release.
Yes, I switch mountpoint references during install and use PARTUUID if installing to a USB drive, otherwise I think it just uses the normal partition references, so it _shouldn't_ make any difference. I'll check it when I convert v1.1.1

sakaki-, 2017-07-21:
Hi - [v.1.1.1](https://github.com/sakaki-/gentoo-on-rpi3-64bit/releases/tag/v1.1.1) is out now. Things to look at that might possibly cause breakage on your side:
* `cmdline.txt` now has `root=PARTUUID=7A0C8BB0-02` in there, in place of `root=mmcblk0p2`;
* similarly, the `autoexpand-root` service now [looks up](https://github.com/sakaki-/rpi3-overlay/commit/f4004c0b16e4b339dc8e1a1514fb4b74bf2afcea) the `root=` entry in `/proc/cmdline`, and then (if a PARTUUID) canonicalizes it; however, this _should_ only happen for the standard sentinel types; the logic should not run with an `autoexpand_root_none` sentinel;
* lastly, the `/etc/fstab` now reads (last three lines):
```
PARTUUID=7a0c8bb0-01	/boot		vfat    defaults          0       2
PARTUUID=7a0c8bb0-02	/		ext4    defaults,noatime  0       1
/var/cache/swap/swap1   none		swap	sw		  0	  0
```

Please let me know if you encounter any issues trying to get the image prepared for your installer, or if there is anything else that you need.

thanks!

procount, 2017-07-21:
Thanks. Downloading now.

procount, 2017-07-23:
**interim report** Just had a chance to test the v1.1.1 PINN installation.
It installed first time and allowed me to enter the correct password for demouser at the cmdline login prompt. However, the rootfs was r/o again, so I'll have to apply some magic....

sakaki-, 2017-07-24:
OK, thanks. Strange about the r/o thing. I have added explicit `fsck -f` checking to the end of the image prepare script, so both partitions should be clean when shipped (and obviously the vanilla image boots up OK from microSD).

One other thing you could try - because the Pi has no RTC, the image uses Gentoo's `swclock` service; this saves the last shutdown time (in `/lib64/rc/cache/shutdowntime`) and restores this in early boot (eventually the `chronyd` NTP client sets the correct time). Large superblock time offsets (last write more than a day in the future, according to the cached shutdown time) will force an fsck to happen during boot. The edits you are making to the two partitions when loop mounted will probably have this effect.

So, you could try touching the file `${YOURLOOPMOUNT}/lib64/rc/cache/shutdowntime` as the last step when editing the image, to see if this makes any difference.

procount, 2017-07-25:
I still haven't quite got to the bottom of why it goes read-only, but I think it is something to do with the timestamps of modified files, as you suggested. If PINN is installing gentoo offline, it may not have a sense of the time either, so it uses a similar fake clock which will again be different (maybe in the past or future relative to the last real gentoo fs access time.) This maybe giving different results from when PINN is connected to a network with the real time.

However, I managed to hand-hold the installation through, and with a few manual tweaks, PINN managed to load Gentoo64 v1.1.1, from a USB boot (not SD card), complete with a nice icon and shiny slideshow!

I'm going to have to take a break from this for a while, but I'll do some more testing and make my scripts more robust to avoid this read-only problem. Then I hope to release it in a week or so (time permitting) :smile:

sakaki-, 2017-07-25:
> Then I hope to release it in a week or so (time permitting) 

Great, thanks for all your time on this. I look forward to trying it out!

> BTW, the cursor no longer disappears when the screen blanks. I guess this is an artefact of the use of the new software cursor?

The software cursor will only kick in if overscan is enabled (and it is now disabled by default in the shipped `config.txt`, unless your installer modifies this). However, there's an upgraded `mesa` library in the 1.1.1 too, so that could explain it.

procount, 2017-07-26:
Arrgghhh! I thought I had cracked it, but even keeping all the timestamps the same, it still went r/o.

sakaki-, 2017-07-26:
Happy to look at an image if you want to upload one somewhere (a dummy project release on GitHub for example)?
Are you still getting the same mount error?
```
[ 8.966408] EXT4-fs (mmcblk0p7): Unrecognized mount option "freely." or missing value.
```

Here's what the `root` init script is doing:
```bash
#!/sbin/openrc-run
# Copyright (c) 2007-2015 The OpenRC Authors.
# See the Authors file at the top-level directory of this distribution and
# https://github.com/OpenRC/openrc/blob/master/AUTHORS
#
# This file is part of OpenRC. It is subject to the license terms in
# the LICENSE file found in the top-level directory of this
# distribution and at https://github.com/OpenRC/openrc/blob/master/LICENSE
# This file may not be copied, modified, propagated, or distributed
# except according to the terms contained in the LICENSE file.

description="Mount the root fs read/write"

depend()
{
	after clock
	need fsck
	keyword -docker -jail -lxc -openvz -prefix -systemd-nspawn -vserver
}

start()
{
	case ",$(fstabinfo -o /)," in
		*,ro,*)
		;;
		*)
			# Check if the rootfs isn't already writable.
			if checkpath -W /; then
				rm -f /fastboot /forcefsck
			else
				ebegin "Remounting root filesystem read/write"
				case "$RC_UNAME" in
					Linux)
						mount -n -o remount,rw /
					;;
					*)
						mount -u -o rw /
					;;
				esac
				eend $? "Root filesystem could not be mounted read/write"
				if [ $?  -eq 0 ]; then
					rm -f /fastboot /forcefsck
				fi
			fi
		;;
	esac

	ebegin "Remounting filesystems"
	local mountpoint
	for mountpoint in $(fstabinfo); do
		case "${mountpoint}" in
			/)
			;;
			/*)
				mountinfo -q "${mountpoint}" && \
					fstabinfo --remount "${mountpoint}"
			;;
		esac
	done
	eend 0
}
```

So once booted r/o, does:
```console
pi64 ~ # mount -n -o remount,rw /
```
work? If not, what is going to `dmesg`?

procount, 2017-07-26:
No it won't mount rw.
If I find time later I will post it up for you to try.

sakaki-, 2017-07-26:
OK thanks, I'll definitely have a look if you do get a chance to post it.

procount, 2017-07-26:
I have uploaded the gentoo64 installation files for you to play with :)

Just follow these instructions (a bit more complicated than usual due to the manual test installation)
1. Get a 16GB or larger SD card (or USB stick if your RPi3 has been programmed for USB boot, but it is slower to boot)
2. Format it to FAT32. I recommend the SDformatter v4 tool on WIndows with size adjustment on to ensure ALL partitions are deleted and the FAT32 partition fills the entire card.
3. Go to https://sourceforge.net/projects/pinn/files/ and download pinn-lite.zip.
4. unzip pinn-lite.zip onto the SD card preserving any folder structure.
5. in the /os folder, create a folder named gentoo64
6. Go to https://sourceforge.net/projects/pinn/files/testing/gentoo64/ and download all files (including the slides_vga folder and it's contents) and put them in the /os/gentoo64 folder on the SD card for local installation. (once released, these files are downloaded directly from the internet)
7. (Optional) in the root of the SD card, edit the recovery.cmdline. This just consists of 1 single line.
7a.  Add the keywords 'ssh vncshare forcetrigger' to this line (no quotes). 
7b.  This enables ssh and vnc support to enable headless use/debugging. forcetrigger forces PINN to display its menu on each boot. For ssh login use 'root/raspberry'. 
7c.  You can also press ctrl-alt-f2 to get to a local debug terminal with the same password. Either is useful for looking at the SD card or gentoo partitions on the RPi whilst in-situ, although you only have a minimal busybox shell.
8. Insert SD card into the RPi3 and boot.
9. You will see gentoo64 as an option in the list. Select the checkbox and select the install button. If the RPI3 is connected to the internet, you will also get a whole lot of other OSes as well. 
10. Once installed click on the OK button. The RPi should boot into Gentoo (if you are lucky)
11. On subsequent reboots, just Exit PINN to boot into gentoo64. (with no forcetrigger option, it will boot gentoo64 after a timeout automatically)

What has PINN done?
The original gentoo64 image was loop mounted.
The 2 partitions have been converted to 2 .tar.xz files, 1 for each partition, with no modifications. It's just a change in format.
It created /dev/mmcblk0p6 for the fat32 boot partition and /dev/mmcblk0p7 for the ext4 root partition.
The 2 tar.xz files are copied onto the respective partitions.
/os/gentoo64/partition_setup.sh is used to modify the gentoo64 files post-installation. Namely, the cmdline.txt file and the etc/fstab
Then it reboots into partition 6 to load gentoo64. The problem then is that it fails to load root r/w for some unknown reason.

Note that PINN will change the diskid and thus the PARTUUIDs used by gentoo64 on a USB stick. SD Cards may still use /dev/mmcblk0pX format.

For more info on how to use PINN and what it does see:
https://github.com/procount/pinn
https://github.com/procount/pinn/blob/master/README_PINN.md
https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version

Enjoy!

sakaki-, 2017-07-27:
OK been through the install process, looks great!
I have reproduced the r/o issue my side. Will dig into it in the next few days.
Thanks, sakaki

sakaki-, 2017-07-28:
Seems to be either the filesystem or tarball contents for partition 7.
If I take your (r/o booting) image and `dd` my image's partition 2 contents over partition 7, then manually edit the `/etc/fstab` entries, touch `/lib64/rc/cache/shutdowntime` and reboot, it all comes up fine.
I will revert your image and try `rm -rf` on partition 7, then `cp -ax`-ing the files from my partition 2 into it next. That should indicate if it is the ext4 filesystem or the root tarball. Will keep you posted.

sakaki-, 2017-07-28:
Doesn't look to be the ext4 filesystem itself - the `cp -ax`'d system (with modified fstab & shutdowntime) boots fine (to r/w & desktop).
So, looks like there may be an issue with the root tarball somehow. I'll unpack it, and do a recursive `diff` against my baseline image's root filesystem next.

sakaki-, 2017-07-28:
Hmmm... the tarball looks almost but not quite the same as the 1.1.1 root... for example, `/usr/include/bsd/sys` seems to be empty in the tarball, but not on 1.1.1. Although I can't understand why that should matter, it's a little odd, so I'm going to try rolling my own tar.xz now and see if PINN will install from that.

sakaki-, 2017-07-28:
OK, interesting. Here are the steps I took to create `root_gen.tar.xz` (the `genpi64.img.xz` image I used is exactly the version posted on GitHub):
```console
root@build1 ~ # cp -iv /home/sakaki/rpi3/1.1.1/genpi64.img.xz .
'/home/sakaki/rpi3/1.1.1/genpi64.img.xz' -> './genpi64.img.xz'
root@build1 ~ # unxz genpi64.img.xz 
root@build1 ~ # kpartx -s -v -a genpi64.img
add map loop1p1 (253:13): 0 129024 linear 7:1 2048
add map loop1p2 (253:14): 0 15200000 linear 7:1 131072
root@build1 ~ # mkdir /mnt/genpi64root
root@build1 ~ # mount -r /dev/mapper/loop1p2 /mnt/genpi64root
root@build1 ~ # du -sh /mnt/genpi64root
6.2G	/mnt/genpi64root
root@build1 ~ # tar cf root_gen.tar -C /mnt/genpi64root .
root@build1 ~ # ls -lh root_gen.tar 
-rw-r--r-- 1 root root 5.4G Jul 28 03:40 root_gen.tar
root@build1 ~ # xz -v -9e -T0 root_gen.tar 
root_gen.tar (1/1)
  100 %     774.8 MiB / 5,461.3 MiB = 0.142    32 MiB/s       2:50             
root@build1 ~ # ls -lh root_gen.tar.xz 
-rw-r--r-- 1 root root 775M Jul 28 03:40 root_gen.tar.xz
```
Copied this to the `os/gentoo64` folder in the PINN first partition's filesystem. Booted PINN and installed Gentoo using this tarball.

Got the r/o issue again.

OK, then after that mounted `/dev/mmcblk0p7` on a PC. `rm -rf *`'d the filesystem (but did not reformat), and then untarred the above `root_gen.tar.xz` into the now-blank root. Fixed up fstab, timestamp, touched `autoexpand_root_none` on `/dev/mmcblk0p6`. Unmounted and booted.

Went through fine this time.

So, looks like:
* the ext4 filesystem is OK
* the tarball contents are OK (at least the one I just created), but
* somehow, the tarball is not getting written correctly by the installer

Should be easy to check this: untar the tarball to HDD, then mount the post-PINN-install `/dev/mmcblk0p7` and recursively diff these two (with `meld` or similar tool). Most likely one or more binaries or libs has been corrupted during the installer untar - although why this should be so I have no idea.

Must run now but perhaps you could run that test your side? Thanks!

procount, 2017-07-28:
Thanks for your investigations. Very similar to what I was doing.
Can't do anything at the moment I'm afraid but will try when I can.
PINN can also install img files so we could also extract the 2 partitions from the original image to 2 separate image files but it will probably require resizing the filesystem to the new disk size on 1st boot.
Dunno why tar is acting to strange. We use bsdtar now if that makes a difference.

sakaki-, 2017-07-28:
**Found it**. Nothing to do with decompression at all, but it was useful to run the recursive diff, as it pinpointed the issue.

Have a look at your patched `/etc/fstab` (after PINN has modified it). On my system at least, looks like:
```bash
# /etc/fstab: static file system information.
#
# noatime turns off atimes for increased performance (atimes normally aren't 
# needed); notail increases performance of ReiserFS (at the expense of storage 
# efficiency).  It's safe to drop the noatime options if you want and to 
/dev/mmcblk0p7  / tail freely.
#
# The root filesystem should have a pass number of either 0 or 1.
# All other filesystems should have a pass number of 0 or greater than 1.
#
# See the manpage fstab(5) for more information.
#

# <fs>   <mountpoint> <type>  <opts>  <dump/pass>

# NOTE: If your BOOT partition is ReiserFS, add the notail option to opts.
/dev/mmcblk0p6  /boot  vfat    defaults          0       2
/dev/mmcblk0p7  /  ext4    defaults,noatime  0       1
/var/cache/swap/swap1   none  swap sw    0   0
```
Look at that sixth line from the top. Explains the "freely." bad option message in `dmesg` too!
Your `partition_setup.sh` sed line:
```bash
sed /tmp/2/etc/fstab -i -e "s|^.* / |${part2}  / |"
```
is a little aggressive, as it doesn't check for "#" characters, maybe something like:
```bash
sed /tmp/2/etc/fstab -i -e "s|^\s*[^#].* / |${part2}  / |"
```
instead?

procount, 2017-07-28:
Excellent! Well done!
Will be able to release when I get back.
Many thanks.

sakaki-, 2017-08-05:
Hi,
I've recently put out v1.1.2, as well as a variant image for those using a Pi-Top (sort of RPi3-based DIY laptop). To save you time, I've created the required tarballs (and tested these with PINN, with the modified `sed` rule), and these are downloadable as follows:
* Standard RPi3 image (v1.1.2):
  * [boot_gen.tar.xz](https://isshoni.org/pinn/boot_gen.tar.xz), MD5 f75761101e3087f7628a4d0135115bba
  * [root_gen.tar.xz](https://isshoni.org/pinn/root_gen.tar.xz), MD5 973ba9d1c6202f5f75b52a6d1a1c3be8
* Pi-Top variant image (NB image names must be changed in json) (v1.1.2):
  * [boot_genpt.tar.xz](https://isshoni.org/pinn/boot_genpt.tar.xz), MD5 712fda8bcf1d00df6e686c7c67a99f21
  * [root_genpt.tar.xz](https://isshoni.org/pinn/root_genpt.tar.xz), MD5 a07d1d4e556160dcb4bf55b7a750b31c

The same slides etc. can be used for both images.

Thanks!
sakaki

PS I have noticed that Xfce shows the 'Settings' filesystem on the desktop (as mountable) but I guess that is to be expected given the NOOBS/PINN layout?

procount, 2017-08-08:
Hi,
That's excellent. Many thanks.
I will copy them onto sourceforge for hosting and let you know when it's done. (Just got back so I've got some sorting out to do first...)
I see you are getting very adept at converting your images to PINN format. So, if you prefer to host them yourself, just let me know. You could then have complete control over the issue of new versions and additional ones, such as for the PiTop. 
Re: Settings, Yes, I guess that is to be expected. Unless you know of a way to hide it.

sakaki-, 2017-08-08:
OK well let's see how this goes and then I may start hosting the files for the next release onwards.
PS according to [this thread](https://ubuntuforums.org/showthread.php?t=1753634#post12043740) if the Settings partition is present in the `/etc/fstab` (as a `noauto` entry) it will not show up on the desktop. So I guess you could `mkdir -p /mnt/Settings` on the root and then >> the necessary line to the end of `/etc/fstab` (in `partition_setup.sh`)?

procount, 2017-08-09:
I've downloaded them now, so will test and host them shortly.

procount, 2017-08-09:
Re; Settings, I will need to have a think about that - whether I can infer the partition device ( /dev/sda5, /dev/mmcblk0p5, or a PARTUUID=xxxx-05) or whether I need to pass it as an additional parameter to the script, so I'll leave it for the moment.

sakaki-, 2017-08-09:
OK no problem. FWIW, I've tried manually setting the line (`/dev/mmcblk0p5 /mnt/Settings ext4 defaults,noatime,noauto 0 0`) into `/etc/fstab`, and it does indeed stop the "Settings" icon showing up greyed out on the desktop.

procount, 2017-08-09:
Hmm.... I installed your v1.1.2 version direct from my SD card with PINN. It booted ok with no r/o issues.
I connected to my wifi ok, but when I clicked on the web browser or the file manager, nothing happened.
So I selected 'restart' and rebooted to PINN and hence to Gentoo again, but this time it only got to the initial black graphical screen with the white mouse cursor and  a tiny white square in the top left corner of the display and then it stopped. After re-installing, the same thing happened.

sakaki-, 2017-08-09:
How strange. OK, I will try re-installing again my side and see if I can reproduce what you are seeing (I had tested before posting but I will double check). This is the vanilla 1.1.2 image, not the Pi-Top one, yes?

procount, 2017-08-09:
Yes. Vanilla as you had posted.
I'm sure you had tested it fully. Maybe something with my setup? I'm using the 7" RPF touchscreen (800x480) and a 32GB SD card. 
Now that I have a good partition_setup script, maybe I will try the v1.1.1 image again for comparison, or build the v1.1.2 tar files myself? But getting late now. Might have to wait til tomorrow.

sakaki-, 2017-08-09:
Good idea - if you get a chance to try 1.1.1 that'd be a useful variable elimination, certainly.
Also, when you have time, would you mind posting your modified `partition_setup.sh` somewhere pls, as I am using a version with my own changes and it'd be sensible to normalize that.
Many thanks!

sakaki-, 2017-08-10:
Just tried fresh install of v1.1.2, MD5's `f75761101e3087f7628a4d0135115bba` and `973ba9d1c6202f5f75b52a6d1a1c3be8` to 32GB card via the PINN you originally directed me to download (with modded `partition_setup.sh` only); was able to boot OK (no r/o) and then connect WiFi, open file browser, and open web browser. Tried on both an computer monitor display and an HDMI TV.
Can't think of anything much that has changed from v1.1.1 that would affect output to your screen res, but maybe you could try an install plugged into a TV or monitor instead, to verify that?
Thanks, sakaki

sakaki-, 2017-08-10:
For reference, here are the `md5sum`s of all the files on the PINN image I am using, prior to first boot (i.e. when the fat partition takes up the full card):

```console
sakaki@build8 ~ $ find /mnt/pinn -type f -exec md5sum {} +
8c49edab0b3872634d268d1603a9d3c4  /mnt/pinn/bcm2708-rpi-0-w.dtb
15b760f5118ba86834f5cf73cefaf8a8  /mnt/pinn/bcm2708-rpi-b-plus.dtb
eca36fe493f3362541f63e853b25516f  /mnt/pinn/bcm2708-rpi-b.dtb
c2411515e29eb0b1f7ed11c9832c1088  /mnt/pinn/bcm2709-rpi-2-b.dtb
74e50e3d6390b5b78db362e1eeb21d04  /mnt/pinn/bcm2710-rpi-3-b.dtb
60009c193abdc91edc6439223266b00d  /mnt/pinn/bootcode.bin
80197f29e05f2187fe3a79c63b3f94f1  /mnt/pinn/BUILD-DATA
c3ee0ceeb8359f40d508e3fb277c2a5f  /mnt/pinn/INSTRUCTIONS-README.txt
6f3431a4192896900951a3274b02d826  /mnt/pinn/defaults/slides/A.png
f0d6cf81b76db1b5efc0eeab6ecd51c0  /mnt/pinn/os/gentoo64/gentoo64.png
2564eb862c8d0e9abf7f2eda7d51c241  /mnt/pinn/os/gentoo64/os_list.json
a27c268f7dd514aa3a4e232bb81fa5d4  /mnt/pinn/os/gentoo64/partitions.json
e1bbfebdceceb3b9a85c64df3faf4ca9  /mnt/pinn/os/gentoo64/marketing.tar
6efa5aea161bd5ad714d064e851b851c  /mnt/pinn/os/gentoo64/os.json
3b472c9d17cc4e902301d715cbfc81dd  /mnt/pinn/os/gentoo64/release_notes.txt
35bed68960c2f65a7b8bb590c2fcce96  /mnt/pinn/os/gentoo64/partition_setup.sh
f75761101e3087f7628a4d0135115bba  /mnt/pinn/os/gentoo64/boot_gen.tar.xz
f70de5e90face97ed36d4adb13a57cc0  /mnt/pinn/os/gentoo64/slides_vga/Slide1.png
23fab5924af48481f9342027fdf8dcb4  /mnt/pinn/os/gentoo64/slides_vga/Slide2.png
f982e89ceabca56d25aaa7f7b535729e  /mnt/pinn/os/gentoo64/slides_vga/Slide3.png
01c78227f3044f72ef2909f72a70e8c4  /mnt/pinn/os/gentoo64/slides_vga/Slide4.png
9dffded709bd6e198be87b9cea8d7a1a  /mnt/pinn/os/gentoo64/slides_vga/Slide5.png
c181f73aeedfa188484c32bf1908572e  /mnt/pinn/os/gentoo64/slides_vga/Slide6.png
973ba9d1c6202f5f75b52a6d1a1c3be8  /mnt/pinn/os/gentoo64/root_gen.tar.xz
448b791c76b481b8250f397690f306f5  /mnt/pinn/overlays/rpi-ft5406.dtbo
a290239cfc535a392900d59e9568fd76  /mnt/pinn/overlays/dht11.dtbo
c565a266717e0594fb84acc2d05d00bc  /mnt/pinn/overlays/iqaudio-dacplus.dtbo
a5c1122fdd4bfb63df4c0b6613d76267  /mnt/pinn/overlays/mmc-overlay.dtb
70e8a67c9e86332e05f4d8f8269359a7  /mnt/pinn/overlays/bmp085_i2c-sensor-overlay.dtb
872f073e4044ac7cf89811f50f64a171  /mnt/pinn/overlays/pitft28-resistive-overlay.dtb
ebabb919a03f788a945de060513bd331  /mnt/pinn/overlays/vc4-kms-v3d.dtbo
fbfe3584aaca14fffef539ec3f074664  /mnt/pinn/overlays/dpi24.dtbo
d37d706e824bc804fdf512f384978933  /mnt/pinn/overlays/at86rf233-overlay.dtb
b13be107d380c1fdf6d1a4538e98362c  /mnt/pinn/overlays/dht11-overlay.dtb
3deab7985101cba215b819ab3cd2bd65  /mnt/pinn/overlays/at86rf233.dtbo
65fd72cc3e008092e67ea291f8d38b5a  /mnt/pinn/overlays/vc4-fkms-v3d.dtbo
7227810e0bac11426ec2f1fb5bd90eab  /mnt/pinn/overlays/spi-bcm2708-overlay.dtb
6d70a24ab287bdf58ddf168292fd92b9  /mnt/pinn/overlays/spi2-3cs.dtbo
df6772b1084a3a39d680c8e8720647a2  /mnt/pinn/overlays/pi3-act-led.dtbo
939443c906ed6e5e9ec4223ad02f8804  /mnt/pinn/overlays/hifiberry-dac-overlay.dtb
6e717584dc6b6efc72a1589201539909  /mnt/pinn/overlays/i2c-rtc.dtbo
7bff9a325a00819223503456fa947a9c  /mnt/pinn/overlays/dwc-otg.dtbo
e9377c23a8169852f050853e17e779cd  /mnt/pinn/overlays/pitft35-resistive.dtbo
3c415b03b102bfe9ed88bf14f22d7a6a  /mnt/pinn/overlays/hy28a.dtbo
70e8a67c9e86332e05f4d8f8269359a7  /mnt/pinn/overlays/bmp085_i2c-sensor.dtbo
895af60c8847f36b92bd511de063649d  /mnt/pinn/overlays/spi2-1cs.dtbo
4dbe6ecaf13243707b27b1e54a321b94  /mnt/pinn/overlays/hifiberry-digi-pro.dtbo
d9cd5417b9b439bed0dde861b1f4a7da  /mnt/pinn/overlays/rpi-ft5406-overlay.dtb
808a6234ca9fbd0c26dc12c023898baa  /mnt/pinn/overlays/vga666-overlay.dtb
15dd29939e6358a5f1295046e91b03e6  /mnt/pinn/overlays/pitft28-capacitive.dtbo
b82cbb15b30f6bcf6fd05ef630c703ce  /mnt/pinn/overlays/dwc2.dtbo
1ad711b67df56c254f522c67a75db669  /mnt/pinn/overlays/i2c-rtc-overlay.dtb
383830fe141c0cb5763e4eeaf22e5393  /mnt/pinn/overlays/rpi-proto.dtbo
143d08fb0929b13e6ccc9bf9c97f3d0a  /mnt/pinn/overlays/smi-nand-overlay.dtb
989a912bcba7a06a769fb7edab77778a  /mnt/pinn/overlays/gpio-ir-overlay.dtb
55d846402c6b97a14eba51d101067be9  /mnt/pinn/overlays/hifiberry-amp.dtbo
f9dd2b3934177d2ea7d7a5a47bbaf1af  /mnt/pinn/overlays/i2c-mux.dtbo
86b12f02e1d138ee6a8a4af4c26c0227  /mnt/pinn/overlays/pi3-miniuart-bt-overlay.dtb
fb78ec271c83dccedd4cac9c939a8bef  /mnt/pinn/overlays/pitft28-capacitive-overlay.dtb
e5939e5e0eb1b264ed6285e1d8a00db5  /mnt/pinn/overlays/hifiberry-amp-overlay.dtb
b7bc0a27f234efbeb05361739c4c18d3  /mnt/pinn/overlays/hifiberry-dac.dtbo
50a0c9d7238553115a684e0374227747  /mnt/pinn/overlays/enc28j60-overlay.dtb
cb2c39fb925d8ee87733e7d39ad18c18  /mnt/pinn/overlays/uart1-overlay.dtb
1ff8cf1d9714d3debd3e5ff18f08d963  /mnt/pinn/overlays/pi3-miniuart-bt.dtbo
d693a0ebe5b4fa3aefb37a94cda9f202  /mnt/pinn/overlays/pitft28-resistive.dtbo
96cf593b3a7b9d1e4e8c34688cfd7f41  /mnt/pinn/overlays/hifiberry-digi.dtbo
[... 151 lines cut ...]

#Update root partition ref in cmdline.txt, keeping the timestamp the same
mv /tmp/1/cmdline.txt /tmp/1/cmdline.bak	#Keeps timestamp
cp /tmp/1/cmdline.bak /tmp/1/cmdline.txt	#timestamp updated
sed /tmp/1/cmdline.txt -i -e "s|root=[^ ]*|root=${part2}|"
touch -r /tmp/1/cmdline.bak /tmp/1/cmdline.txt	#restore timestamp
rm /tmp/1/cmdline.bak

#Update partition refs in fstab, keeping the timestamp the same
mv /tmp/2/etc/fstab /tmp/2/etc/fstab.bak	#Keeps timestamp
cp /tmp/2/etc/fstab.bak  /tmp/2/etc/fstab	#timestamp updated
[... cut ...] /master/boot/overlays/READMEhttps://github.com/raspberrypi/firmware/blob/master/boot/overlays/README| |g"
sed /tmp/2/etc/fstab -i -e "s|^[^#].* / |${part2}  / |"
sed /tmp/2/etc/fstab -i -e "s|^[^#].* /boot |${part1}  /boot |"
touch -r /tmp/2/etc/fstab.bak /tmp/2/etc/fstab	#restore timestamp
rm /tmp/2/etc/fstab.bak

#Prevent root partition expansion - already doen by PINN
mv /tmp/1/autoexpand_root_partition /tmp/1/autoexpand_root_none #Keeps timestamp

umount /tmp/1
umount /tmp/2
```

hth, sakaki

procount, 2017-08-10:
I created my own tar files from v1.1.2 and installed via PINN using the 7" RPF touchscreen at 800x480 resolution. It worked fine. Then I clicked on "Demo User" in top right, "logout - restart". When I got through to Gentoo the 2nd time, this is where the problems started. Once, the background wallpaper did not appear and it remained black but with desktop icons. But mostly, the top status bar and bottom launcher window were absent (see attached image)
[image]
Restarting with HDMI connected seems to restore the missing taskbars to the desktop.
My results are not conclusive yet, but it seems there is a problem saving the screen size (?) on the first RESTART when using a small screen (or the touchscreen?). It doesn't seem to happen if you use HDMI at 1650x1050 for example, or if you use SHUTDOWN. Needs more testing for a definitive answer, but I think this is a separate issue that is unrelated to the PINN installation (or maybe I am doing something wrong)

Anyhow, I will find some time to post my PINN gentoo64 installation and you can verify my files. Then I think we're good to go!

re. partition_setup.sh: I removed all my unnecessary timestamping nonsense. I was using your previous sed arguments, but I see you have updated them slightly, so I will do the same.
What is the `sed /tmp/2/etc/fstab -i -e "s|\thttps://github.com/raspberrypi/firmware/blob/master/boot/overlays/READMEhttps://github.com/raspberrypi/firmware/blob/master/boot/overlays/README| |g"` line for?

sakaki-, 2017-08-10:
Thanks, that's interesting. When you get the situation in the screenshot above, you could try running `xfce4-panel&` in a terminal (as demouser) and see if it produces any interesting error output (or whether this in fact restores the panels). It could be a timing thing. There's a file `/usr/bin/xfce4-fixups-rpi3` which reloads the deskop etc. (to get around the black background issue, which is a known Xfce4 problem); this reload happens after a certain delay from autostart, which may be too short on small displays, not sure. Anyway as you say needs more testing - larger displays seem to work OK at the moment.

That strange `sed` line you mentioned can be ignored - it was left over from a test.

When you have an uploaded image, I will be happy to download and try it out.

Best, sakaki

procount, 2017-08-11:
I've uploaded my installation. Change recovery.cmdline to the following to install gentoo from the web:

`quiet ssh forcetrigger vncshare ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline alt_image_source=http://raw.githubusercontent.com/procount/pinn-os/master/os/os_gentoo.json`

Also rename /os/gentoo64/os.json to hide the existing installation on the SD card (rather than deleting it).
Then you can reinstall gentoo64 from the web. Alternatively, you can download the installation files for your SD card from `https://sourceforge.net/projects/pinn/files/host/gentoo64/`

I just installed it from the web
After first boot, I SHUTDOWN, then rebooted and I got the missing panels again. I tried `xfce4-panel&` which resulted in `xfce4-panel: There is already a running instance`

I did `sudo reboot` and on the next boot it was back to normal again. Very odd....

sakaki-, 2017-08-11:
Thanks, downloading it now.
If you get a chance, it'd be very useful if you could run the following in the 'no panels' situation (as per your screenshot above; run as the normal user):
```bash
demouser@pi64 ~ $ xfconf-query --channel xfwm4 --property /general/setup_complete
demouser@pi64 ~ $ xfconf-query --channel xfce4-panel --property /general/setup_complete
demouser@pi64 ~ $ xfce4-panel --restart
```
The first two of these are sentinel properties written by the 'xfce4-fixups-rpi3' script ([here](https://github.com/sakaki-/rpi3-overlay/blob/master/xfce-extra/xfce4-fixups-rpi3/files/xfce4-fixups-rpi3-1)); it may be that this is not working correctly on small screens and it'd be very helpful to know this (it is a change in 1.1.2). The last should make the panels re-appear assuming that `xfce4-panel` is running already (which your previous response suggests it is). Many thanks!

sakaki-, 2017-08-11:
Hi,

OK downloaded and booted fine; on a system with a standard sized screen at least was able to reboot etc. without any obvious panel problems.

No obvious issues in `dmesg` `/var/log/messages` or `/var/log/rc.log` (other than that there is now a > 1day jump in last write time on root, which triggers an automatic fixup on first boot, not a big issue and I guess related to the timestamp stuff you removed).

Only thing on the desktop is the dimmed out "Settings" icon which may confuse some people I suppose (they may be looking for system settings, and double click it, mounting the PINN partition). This is a PINN/Xfce4-specific thing so in time perhaps we should add something to address it (the `/etc/fstab` tweak will do the job).

**Many thanks** for putting this together! The standard image is good to ship from my perspective (I would like to get to the bottom of the small screen problem, but I am confident that has nothing to do with PINN, so it shouldn't hold up getting the v1.1.2 out).

If you have the Pi-Top variant hosted somewhere, I am happy to try that also.

[121 more replies]

closed 2017-10-02 by sakaki-