sakaki-/gentoo-on-rpi-64bit #16: Sluggish XFCE

olle2, 2017-07-19:
Hi Sakaki, thanks for the new toy! I have tested https://github.com/sakaki-/gentoo-on-rpi3-64bit/releases/latest and it certainly works. I think however that its a little bit slow/sluggish, maybe because of XFCE. 
I have tested  https://github.com/bamarni/pi64/releases and https://files.devuan.org/devuan_jessie/embedded/ on the same pi and sdcard and they seem to run faster.

So I need your help to change from XFCE to lxde or lxqt.

Thanks!

sakaki-, 2017-07-19:
What particularly is it that appears to run slowly; is it the desktop experience you mean (moving windows round the screen, menu redraws etc)? Or do you mean that apps themselves appear to run slower?

If the former, have you tried turning off display compositing? Select <kbd>Applications</kbd>&rarr;<kbd>Settings</kbd>&rarr;<kbd>Window Manager Tweaks</kbd>, select the `Compositor` tab, and untick `Enable display compositing`.

Of course, it is not difficult to switch to a different desktop - Gentoo has both `lxde` and `lxqt` in tree. You just need to unmerge the current `xfce4-meta` package and emerge either `lxde-meta` or `lxqt-meta` and then patch whatever is broken (e.g. `lightdm`). Doable but in the sort term I don't have bandwidth to support two variants like this. However, if you want to give it a go, please let me know if you experience significant performance improvement.

Best, sakaki

olle2, 2017-07-19:
Thanks for your answer! Its the apps themselves running slowly (particularly VLC and FF) but I'll try the display composting, and maybe later the unmerging and emerging of desktops. New to Gentoo, so no guaranties!

closed 2017-07-19 by olle2