sakaki-/gentoo-on-rpi-64bit #20: Kodi dependencies

karl6994, 2017-08-19:
Hi! I'm really new to ARM processors & Gentoo, when I last tried your Arm64 Rpi3 build, I wasn't able to install Kodi - there was a dependency that wasn't working, I think it was java/icedtea-bin or something like that.. I'm just curious if anyone has been able to make Kodi work!! That would have me SOLD on this distro if Kodi worked!! I will attempt to use your build again if Kodi is working, thanks for your great contribution to the Linux community.

sakaki-, 2017-08-19:
Not present yet, but icedtea is next on my hit list. Others (NeddySeagoon) have got it to build on `arm64`, so should be in shortly. Watch this space ^-^

sakaki-, 2017-08-23:
Hi again,

I've just added `kodi-17.4_rc1` to the weekly autobuild / binhost (and `icedtea-3.5.1` & `icedtea-7.2.6.11` too; while these are _not_ needed to install or use the `kodi-17.4_rc1` binary package per the following instructions, you'll need one of them to _build_ `kodi-9999` yourself).

To try this out, download and boot a fresh copy of the [current 1.1.2 image](https://github.com/sakaki-/gentoo-on-rpi3-64bit#downloadlinks). Ensure your RPi3 has network connectivity (make sure you can ping out), and wait (important!) for the desktop clock to autosync to the correct time (should happen within 5 minutes or so).

Then, open a terminal, and run:
```console
demouser@pi64 ~ $ sudo genup
```
Let this run to completion, it should take no more than 2 hours provided your Internet connection is reasonable, and will fully update the image using binary packages from the binhost.

Once this completes successfully (you see the message `* All done - your system is now up-to-date!`), reboot, then open a terminal, and run:
```console
demouser@pi64 ~ $ sudo emerge -v kodi
```
to pull in `kodi-17.4_rc1` and dependencies from the binhost. It shouldn't take long. That's it! You can then try it out (<kbd>Applications</kbd>&rarr;<kbd>Multimedia</kbd>&rarr;<kbd>Kodi</kbd>).

Enjoy ^-^

Important: bear in mind when playing videos with kodi, that currently MMAL and OpenMAX hardware accelerated decoding are _unavailable_ in 64-bit mode (see my post [here](https://forums.gentoo.org/viewtopic-p-8106974.html#8106974)); this is a generic issue for all 64-bit RPi3 distros at the moment, not just Gentoo. So performance might not be as good as you would like (it will be using software video decoding). Be sure to turn off display-manager compositing when playing full-screen video (via <kbd>Applications</kbd>&rarr;<kbd>Settings</kbd>&rarr;<kbd>Window Manager Tweaks</kbd>, <kbd>Compositor</kbd> tab). A decent heatsink may also be required to prevent thermal throttling (indicated by a "red thermometer" icon on the top-right of your screen) during playback.

PS it is possible that some kodi features may not be working correctly, as I don't use this app myself and so have only done some very preliminary testing. But, quite a few people have requested it so, here it is!

PPS to install Java (needed if you want to build the 'live' (aka 9999) kodi ebuild), issue:
```console
demouser@pi64 ~ $ sudo emerge -v icedtea:8
demouser@pi64 ~ $ sudo java-config --set-system-vm icedtea-8
```

But I would definitely recommend trying the binary package route for kodi first, before attempting a 9999 build. It will at least give you a known baseline to compare against.

Any issues please let me know! Best, sakaki

karl6994, 2017-08-23:
WOW! Thank you SO MUCH!! I will finally be able to put my Raspberry Pi 3B to good use with arm64! :) 
Qupzilla & Clementine would also be super cool :d  
Qupzilla is a really good lightweight browser.. these things may already work, I don't know. I'm going to attempt this now! Thank you!

sakaki-, 2017-09-17:
Release v1.1.3 of the image is now out, and contains Kodi and Clementine pre-installed.
Had a look at Qupzilla and got it to build, but not run, on `arm64` (segfaults) so it didn't make the cut for now.
Best, sakaki