sakaki-/gentoo-on-rpi-64bit #21: Motion, MotionEye, Create_AP

karl69942, 2017-10-01:
Hi Sakaki, my other account got messed up so I can't reply to it :( 
However I did get emails about Kodi+Clementine working & I think you said you were still trying to get Qupzilla to work, thank you for that! 
I was wondering if you could get Motion, MotionEye & Create_AP to work on your Gentoo build!
However, I am not entirely sure if they work or not on your build already, so, yea.. 
www.github.com/oblique/create_ap
www.github.com/ccrisan/motioneye
www.github.com/motion-project/motion

kenny-wettlaufer, 2017-11-26:
I got bored & decided to give your Gentoo build a shot again, glad to see LXDE & some of the other things that weren't working last time are working now! How do I get MotionEye to work with OpenRC though? Any ideas? I tried to use the sysvinit script to see if it'd work with OpenRC but unfortunately it didn't.
This is the error I got:

/etc/init.d/motioneye.init start
/etc/init.d/motioneye.init: line 23: /lib/lsb/init-functions: No such file or directory

sakaki-, 2019-08-17:
Old issue, closing now in pre 1.5.0 housecleaning

These packages may work under the latest Gentoo, but I haven't had a chance to look at them (might just need keywording).
Please re-open if you still would like me to look at this.
Best, sakaki

closed 2019-08-17 by sakaki-