sakaki-/gentoo-on-rpi-64bit #22: Any chance to get a 4k resolution?

danielmorlock, 2017-10-12:
It looks like the Raspberry Pi 3 may be powerful enough to output 4K video, but xrandr only shows max. 1920x1080. Any change to get a higher resolution?

```
$ XAUTHORITY=/home/demouser/.Xauthority DISPLAY=:1 xrandr     
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 2048 x 2048
HDMI-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   1920x1080     60.00* 
```

sakaki-, 2017-10-15:
Haven't really looked at getting 4K - have you tried the suggestions in [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=79330)?

sakaki-, 2019-08-17:
Old issue, closing now in pre 1.5.0 housecleaning

closed 2019-08-17 by sakaki-