sakaki-/gentoo-on-rpi-64bit #24: Problem with crossdev building

afonsoklein, 2017-11-06:
Hi! Thanks for your fine work! I'm a complete newcomer to the Gentoo distribution, but very interested in the Raspberry Pi 3 port. The crossdev facility is also very interesting, and I managed to have a minimal Gentoo AMD64 running under virtualbox. When following your cross-compilation setup instructions, I got these messages:

crossdev -v -t aarch64-unknown-linux-gnu --genv 'USE="cxx multilib fortran -mudflap nls openmp -sanitize"'
.
.
.
Calculating dependencies  ... done!
[ebuild     U ~] sys-libs/binutils-libs-2.29.1-r1:0/2.29.1::gentoo [2.28.1:0/2.28.1::gentoo] USE="nls -64-bit-bfd -multitarget -static-libs" ABI_X86="(64) -32 (-x32)" 28,459 KiB
[ebuild  N    ~] cross-aarch64-unknown-linux-gnu/binutils-2.29.1-r1:2.29.1::crossdev  USE="cxx nls -multitarget -static-libs {-test} -vanilla" 21 KiB

Total: 2 packages (1 upgrade, 1 new), Size of downloads: 28,480 KiB

The following keyword changes are necessary to proceed:
 (see "package.accept_keywords" in the portage(5) man page for more details)
# required by cross-aarch64-unknown-linux-gnu/binutils-2.29.1-r1::crossdev
# required by cross-aarch64-unknown-linux-gnu/binutils (argument)
=sys-libs/binutils-libs-2.29.1-r1 ~amd64

Use --autounmask-write to write changes to config files (honoring
CONFIG_PROTECT). Carefully examine the list of proposed changes,
paying special attention to mask or keyword changes that may expose
experimental or unstable packages.

 * binutils failed :(
 *
 * If you file a bug, please attach the following logfiles:
 * /var/log/portage/cross-aarch64-unknown-linux-gnu-info.log
 * /var/log/portage/cross-aarch64-unknown-linux-gnu-binutils.log.xz
 * /var/tmp/portage/cross-aarch64-unknown-linux-gnu/binutils*/temp/binutils-config.logs.tar.xz

Would you please help me solve this puzzle?

Thanks!

rjhwelsh, 2017-11-06:
You require the ~amd64 keyword to be set for binutils to proceed.

As root run the following to set the keyword.
`echo "=sys-libs/binutils-libs-2.29.1-r1 ~amd64" >> /etc/portage/package.accept_keywords`

Then retry your command.
`crossdev -v -t aarch64-unknown-linux-gnu --genv 'USE="cxx multilib fortran -mudflap nls openmp -sanitize"'`

See [ACCEPT_KEYWORDS](https://wiki.gentoo.org/wiki/ACCEPT_KEYWORDS) for more info. ~AMD64 is the unstable set of packages for AMD64; almost all of the arm64 stuff is unstable atm.

Good luck!

afonsoklein, 2017-11-06:
Thanks a lot, worked like a charm! I'm realizing that Gentoo is similar to some sportive cars: when you buy one, you should hire the mechanic that assembled the engine. In my case, I´ll have to turn myself into that mechanic! Let´s cross-compile a kernel and read ACCEPT_KEYWORDS in the meantime !!! 8-)

closed 2017-11-06 by afonsoklein