sakaki-/gentoo-on-rpi-64bit #25: F2FS over EXT4

fordfrog, 2017-11-15:
Hi, it's just a question. Gentoo wiki documentation on [Raspberry Pi](https://wiki.gentoo.org/wiki/Raspberry_Pi#Partitioning) mentions that [F2FS](https://en.wikipedia.org/wiki/F2FS) is suitable filesystem for SD cards. Are there really any benefits to using F2FS over EXT4 for this project?

sakaki-, 2017-11-15:
It may help extend the lifetime of the SD card somewhat. Not sure it will be much faster though with modern kernels, see e.g. [this set of benchmarks](https://www.phoronix.com/scan.php?page=article&item=linux-412-fs&num=2). `ext4` is certainly more proven, which is why I used it in the project originally.
But, if you try it and find a big improvement, please let me know!

kenny-wettlaufer, 2017-11-24:
F2FS is made for flash memory, I hear its really good for SSD's and cards. 
I think F2FS has great potential, too bad Etcher doesn't give me an option to choose the format.

sakaki-, 2018-04-05:
OK, perhaps I should take a closer look at this. Now the v1.2.1 image is out, I'll do some comparative benchmarking of F2FS vs ext4 for the root filesystem, under realistic loads, and see how it performs.

sakaki-, 2018-04-08:
I have a booting system with an F2FS root working now, 64MiB-aligned sectors etc. Seems to work reasonably well, going to do some benchmarks next.
One annoying thing is that `resize.f2fs` does not appear to support online resizing, as `resize2fs` does. That would mean using an initramfs during boot to allow the root fs to be expanded.

sedlund, 2019-07-16:
Given that The Pi3 would limit the speed of the SD.  I really don't see any advantage to switching.  And as mentioned you loose online resize.

sakaki-, 2019-08-17:
Old issue, closing now in pre 1.5.0 housecleaning
`f2fs` is built into the modern kernels, so if anyone has relevant benchmarking to report that touches on this question, please feel free to report here.
Best, sakaki

closed 2019-08-17 by sakaki-