sakaki-/gentoo-on-rpi-64bit #26: MotionEye with OpenRC

kenny-wettlaufer, 2017-11-26:
I got bored & decided to give your Gentoo build a shot again, glad to see LXDE & some of the other things that weren't working last time are working now! How do I get MotionEye to work with OpenRC though? Any ideas? I tried to use the sysvinit script to see if it'd work with OpenRC but unfortunately it didn't.
This is the error I got when trying out motioneye:

/etc/init.d/motioneye.init start
/etc/init.d/motioneye.init: line 23: /lib/lsb/init-functions: No such file or directory

sakaki-, 2017-11-27:
That file (`/lib/lsb/init-functions`) isn't present in OpenRC; you'll need to adapt the init script (it looks reasonably straightforward). Begin by commenting out line 23 in `/etc/init.d/motioneye.init` and trying again. What error do you get then?

sakaki-, 2017-11-27:
Based on the `systemd` unit, you should at least be able to try it out by issuing (as root):
```console
# /usr/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
```
although this won't give you `start-stop-daemon`'s PID control etc.