sakaki-/gentoo-on-rpi-64bit #28: Use virtuals in dev-embedded/rpi3-64bit-meta

V10lator, 2017-12-02:
Your meta package makes it hard to switch to systemd. It would be way more simple if you would switch some of the dependencies to virtuals, i.e. change

sys-fs/eudev

to

virtual/udev

sakaki-, 2017-12-02:
There are a lot of OpenRC packages currently in the [rpi3 overlay](https://github.com/sakaki-/rpi3-overlay) used by the image; without these, the system as currently presented won't work. I have nothing against `systemd` though, and may make a variant targeting this in future, but there is no easy fix for the current image (unless you want to drop to stage-3 equivalent functionality, in which case, just start from the stage 3 ^-^). Closing.

closed 2017-12-02 by sakaki-

V10lator, 2017-12-12:
The system works as currently presented with minor tweaks (writing systemd units as replacements for the init scripts) just fine. Also don't you think it's impossible to solve bug #10 without switching to virtuals (you also force one specific cron service, one specific NTP daemon and so on) ?

Here some unit files, in case you'll need them:

```
[Unit]
Description = Bluetooth attach
After = local-fs.target bluetooth.service

[Service]
Type = forking
ExecStart = /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

[Install]
WantedBy = multi-user.target
```

```
[Unit]
Description=Switch CPU to ondemand
After=local-fs.target

[Service]
Type=simple
ExecStart=/usr/local/sbin/rpi3-activate-ondemand.sh

[Install]
WantedBy=multi-user.target
```

And /usr/local/sbin/rpi3-activate-ondemand.sh (you could probably also wrap this around your init script somehow as the script is just a simplified version of your init script anyway) :

```
#!/bin/sh

SCPU=/sys/devices/system/cpu
SCPU_GOV=${SCPU}/cpu0/cpufreq/scaling_governor
SCPU_OD=${SCPU}/cpufreq/ondemand

if [ -e "${SCPU_GOV}" ]; then
        echo "ondemand" > "${SCPU_GOV}"
        echo 50 > ${SCPU_OD}/up_threshold
        echo 100000 > ${SCPU_OD}/sampling_rate
        echo 50 > ${SCPU_OD}/sampling_down_factor
        echo 1 > ${SCPU_OD}/io_is_busy
fi
```

I didn't port the third init script as I don't need it but it should be fairly simple, too.