sakaki-/gentoo-on-rpi-64bit #32: RPi3 serial number (do I need an MPEG licences key?)

fordfrog, 2017-12-12:
I used to use RPi2 and there it was needed to buy MPEG license key that was provided based on the serial number of the device. I thought the same applies for RPi3 but I cannot get the serial number from cpuinfo:

```
# cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
```
So, is MPEG license key needed for RPi3 and if so, where do I find the serial number?

fordfrog, 2017-12-12:
@sakaki- thank you for the information.

Avamander, 2019-01-18:
@sakaki- any updates on that? Any other issue trackers I could ping?

sakaki-, 2019-01-19:
Hi @Avamander,

I think that's still the case, unfortunately. You could bump [this issue thread](https://github.com/raspberrypi/userland/issues/460#issuecomment-383151653) I guess, or better perhaps, post an issue in the [RPi3 forums](https://www.raspberrypi.org/forums). 6by9 is an active RPF dev. You could also contact Eric Anholt (https://github.com/anholt) himself.

sakaki-, 2019-03-07:
As of the 1.4.0 release of the image, **the RPi3's video codecs are now available**, via [v4l2 m2m](https://events.static.linuxfound.org/images/stories/pdf/lceu2012_debski.pdf) ^-^

sakaki-, 2019-03-07:
Incidentally, you can get the serial number in 64-bit mode by issuing:
```console
demouser@pi64 ~ # cat /proc/device-tree/serial-number
```

sakaki-, 2019-04-25:
Closing now, since the mpeg2 codec (as noted [above](https://github.com/sakaki-/gentoo-on-rpi3-64bit/issues/32#issuecomment-470368264)) is now available, via v4l2 m2m.

closed 2019-04-25 by sakaki-