sakaki-/gentoo-on-rpi-64bit #33: Consider enabling CONFIG_MEDIA_CEC_SUPPORT in kernel

fordfrog, 2017-12-14:
Looking at https://github.com/sakaki-/gentoo-on-rpi3-64bit/blob/master/configs/bcmrpi3_defconfig#L2908 CONFIG_MEDIA_CEC_SUPPORT is disabled which effectively disables support for TV remotes in Kodi so if one wants to use RPi3 as media box for TV, he/she has to recompile kernel to add support for HDMI CEC. Also USE flag "cec" is needed on media-tv/kodi to make it work. That way one could use this image also as Kodi media box with no-time, compared to recompilation of kernel and kodi.
Also, it would be useful to provide current kernel config in the image (/proc/config.gz).

sakaki-, 2017-12-14:
You can get the config by issuing (as root):
```console
modprobe configs
```
after which `/proc/config.gz` is visible. It's done this way to save memory.

There are a few extra kernel settings that'd be useful to have. The issue is that the kernel is slaved to upstream's `bcmrpi3_defconfig`, for compatibility. I might add a second (binary) kernel package with more options enabled in future; this wouldn't be too expensive to support; if I do, I'll make sure `CONFIG_MEDIA_CEC_SUPPORT` is enabled.

dracwyrm, 2017-12-14:
Maybe file a bug report upstream for them to add it to the official supported config, then everyone would benefit.

fordfrog, 2017-12-15:
I just tried latest OSMC to check whether my TV really supports HDMI CEC which RPi can talk to and it works without issue. But when running Gentoo on RPi, i have Inputs -> Peripherals disabled in Kodi and cec-client shows no devices (I suppose that is the reason why Peripherals in Kodi are disabled). OSMC also has kernel config in /boot/ dir (it has kernel 4.9.29-10-osmc) and that has CONFIG_MEDIA_CEC disabled so it seems it is not needed for working HDMI CEC. But then I'm clueless about the reason why on Gentoo I cannot make HDMI CEC to work.

fordfrog, 2017-12-15:
looking at the dev-libs/libcec ebuild, it needs to link against media-libs/raspberrypi-userland to work on RPi, but this feature is disabled by profile i guess.

sakaki-, 2018-06-10:
Hi, I have now released [bcmrpi3-kernel-bis](https://github.com/sakaki-/bcmrpi3-kernel-bis) as the new default kernel autobuild for the image.So, if you'd still like `CONFIG_MEDIA_CEC_SUPPORT` enabled, please open a PR on that project. Thanks!
Closing.

closed 2018-06-10 by sakaki-