sakaki-/gentoo-on-rpi-64bit #35: Are 17 profiles safe?

kde99, 2017-12-31:
Well since the 17 profiles are out for along time (at least on x86). Is it safe to use the 17 profiles just by themselves or do I need to modify them manually? Well I just installed the ISO but I don't wanna recompile a lot on my pi and the 17 profiles comes with a rebuild. Also is it safe to use systemd instead of initd? Or would it be better if I tried bootstrapping a new gentoo from a tarball since I wanna use this as a server (without desktop, managed through ssh)?

yourpleasure, 2018-01-01:
I used 17 profiles, but i have recompile everything myself. By the way, my system is all compiled by myself and not use the ISO by sakaki-. And i use systemd.

kde99, 2018-01-01:
@yourpleasure How did you do it? The tutorial on the wiki doesn't boot on my pi :(

sakaki-, 2018-01-02:
@kde99 -
You can use the 17 profiles, but not with the binhost currently, as all the packages there are 13 profile-based and so are not built with `pie`. I plan to release a 17 based version of the image once I have it stabilized my end.
As to `systemd`, yes this is also possible, but if you want to modify the image to use it you'll need to provide units for a few of the existing OpenRC startup services.

yourpleasure, 2018-01-09:
@kde99 I want to know which part you have problem on.

yourpleasure, 2018-01-09:
@kde99 Sorry to response so late.
I want to know which part you have program on.
If you have problem to use `systemd`, you may need to check some kernel configuration and recompile your kernel. You can read [systemd - Gentoo Wiki](https://wiki.gentoo.org/wiki/Systemd).
After `emerge` `systemd`, you should edit `/boot/cmdline.txt` and change `init` to `init=/usr/lib/systemd/systemd`.
If you have further problem, you can ask me again.

sakaki-, 2018-01-25:
I've just pushed v1.2.0 of the image, which is built on the 17.0 profile.

sakaki-, 2018-04-04:
Closing, the 17.0 profile does not seem to have any pie-specific issues (other than compiling the kernel, due to a known `binutils` problem #39). You should be fine to switch to systemd if you want; I just don't have the bandwidth to support / test two different init systems on an image this size, so going to stick with OpenRC for now.

closed 2018-04-04 by sakaki-