sakaki-/gentoo-on-rpi-64bit #37: The green state LED is not working

3PO, 2018-01-19:
Usually the green onboard LED should show the SD card activity, but with this distro the LED is allways off.

This is the issue from gentoo-on-rpi3-64bit:

    pi64 ~ # ls -r /sys/class/leds/*
    ls: cannot access '/sys/class/leds/*': No such file or directory
    pi64 ~ #

And this is from raspian:

    raspberrypi ~ # ls -r /sys/class/leds/*
    /sys/class/leds/led1:
    uevent  trigger  subsystem  power  max_brightness  device  brightness  ..  .
    
    /sys/class/leds/led0:
    uevent  trigger  subsystem  power  max_brightness  device  brightness  ..  .
    raspberrypi ~ #

3PO, 2018-01-26:
The problem still exists in the new version 1.2.0.

sakaki-, 2018-01-26:
Needs `CONFIG_BCM_VCIO` and `CONFIG_GPIO_BCM_EXP` available and set in the kernel (see e.g. https://github.com/bamarni/pi64/issues/6); this was added in the 4.11.y branch (https://github.com/raspberrypi/linux/commit/3bd444137771bce87ca776393a47be213b67a0e7), but the currently shipped kernel with the image is 4.10.y (or 4.9.y) for vc4 accelerated video driver support / stability. Will have a look at booting 4.15 sometime soon though, now 1.2.0 release is out.

yrabbit, 2018-03-14:
It indeed does work with newer kernels.
=============
rabbit@pi64 ~/kbuild/linux % uname -a
Linux pi64 4.16.0-rc2-v8+ #1 SMP PREEMPT Mon Feb 26 07:25:48 GMT 2018 aarch64 GNU/Linux
ROOT@pi64 ~ # ls -r /sys/class/leds/*
/sys/class/leds/led1@  /sys/class/leds/led0@
=============

sakaki-, 2018-03-24:
I've [unmasked](https://github.com/sakaki-/rpi3-overlay/commit/829c74920fb733cbc77705471c32f69f47154a1f) 4.14* now, if you `emaint sync --repo rpi` and then update `bcmrpi3-kernel-bin` and reboot, you should have a working green LED ^-^

sakaki-, 2018-04-04:
v1.2.1 is out now - supports the RPi3 B+ (as well as B), uses kernel 4.14, and... the green LED finally blinks ^-^

yrabbit, 2018-04-04:
Very good :)

closed 2018-04-08 by sakaki-