sakaki-/gentoo-on-rpi-64bit #39: Compile kernel from source error (pie?)

skulko, 2018-03-08:
I tried to compile kernel from source following the tutorial ("Optional: Compiling a Kernel from Source"). However it always fails with errors like these: 
`  MODPOST vmlinux.o
ld: arch/arm64/kernel/head.o: relocation R_AARCH64_ABS32 against `_kernel_offset_le_lo32' can not be used when making a shared object
ld: init/main.o: relocation R_AARCH64_ABS32 against `__crc_system_state' can not be used when making a shared object
ld: init/version.o: relocation R_AARCH64_ABS32 against `__crc_init_uts_ns' can not be used when making a shared object
ld: init/do_mounts.o: relocation R_AARCH64_ABS32 against `__crc_name_to_dev_t' can not be used when making a shared object
ld: init/init_task.o: relocation R_AARCH64_ABS32 against `__crc_init_task' can not be used when making a shared object

.....

__efistub_sort.c:(.init___ksymtab+sort+0x0): dangerous relocation: unsupported relocation
__efistub_sort.c:(.init___ksymtab+sort+0x8): dangerous relocation: unsupported relocation
ld: warning: creating a DT_TEXTREL in a shared object.
make: *** [Makefile:1010: vmlinux] Error 1`

I read that PIE might be the problem... but system should be pie compatible?!
 I have tried different versions 4.9.y, 4.14.y, 4.16.y, all with the same result..

sakaki-, 2018-03-09:
If you are running the 1.2.0 release of the gentoo-on-rpi3-64bit image (or have upgraded to same) then yes, your system uses a 17.0 profile , which has the `pie` flag set by default for `gcc`. Which `gcc` version are you using (please post the output of `gcc-config --list-profiles`)? I have switched (back) to using the 6.4.0 version for kernel compilation on `arm64`, as I've found the 7.x variants too unstable.

skulko, 2018-03-09:
@sakaki- : yes I got the 1.2.0 release and gcc was downgraded to gcc 6.4.0(-r1) when I did 
`echo "sys-devel/gcc -~arm64" >> /etc/portage/package.accept_keywords/gcc`
I had to compile it locally (not from the binhost) due to use flag mismatch.. but now I am using gcc-6.4.0-r1. 
It is also working well for other packages from the repository. Only kernel and net-libs/nodejs (which freezes terminal and makes it impossible to login via ssh) are making trouble so far...

skulko, 2018-03-12:
dev-libs/elfutils was missing when I ran emerge =raspberrypi-sources-4.14.9999 but I get the same errors as before:
`*** can not be used when making a shared object
***  dangerous relocation: unsupported relocation
ld: warning: creating a DT_TEXTREL in a shared object.
make: *** [Makefile:1010: vmlinux] Error 1`

sakaki-, 2018-03-16:
I normally build `arm64` kernels using `crossdev` on my PC, but I'll have a look at the build process natively on the RPi3 next week to see if I can reproduce this.

sakaki-, 2018-03-22:
I have reproduced this error now with a local-only build. It doesn't happen on a crossdev-only build (same compiler, same USE flags), so possibly something to do with `binutils` or similar, rather than `gcc` per se. Investigating.

sakaki-, 2018-03-23:
Yep, `binutils` - please see https://forums.gentoo.org/viewtopic-p-8199952.html#8199952

skulko, 2018-03-27:
Thanks a lot for your investigating!

sakaki-, 2018-05-15:
To recap, the "dangerous relocation" problem is a [known issue](https://forums.gentoo.org/viewtopic-p-8199952.html#8199952) with the version of `binutils` shipped on the image (2.30). It is fixed upstream in a later binutils release (upstream patch included in 2.30-r2); and (if using `gcc-6.4.0` or `gcc-6.4.0-r1`) the prior version of binutils (2.29.1-r1) also works. So, to compile a kernel locally on your RPi3, you can either:
* use `gcc-6.4.0-r1` and `binutils-2.29.1-r1` (see my instructions [here](https://forums.gentoo.org/viewtopic-p-8217992.html#8217992)); _or_
* use `gcc-7.3.0-r1` and `binutils-2.30-r2`; to do so, start from a _fresh_ copy of the image, then run `genup` (as root) which will update your `gcc` and `binutils` to these versions,  and then perform your normal kernel build steps (`make distclean` etc.) as described [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit#kernelbuild) (but _ignore_ the steps before the sentence "Now, suppose you wish to build the most modern version of the 4.14.y kernel...")

Apologies for the inconvenience, closing this issue as I believe it is fixed now (and the documentation [has been modified](https://github.com/sakaki-/gentoo-on-rpi3-64bit/commit/cd59b316ca4ca51b295f073240546436fe0b6695) to recommend stable binutils as well as stable gcc, which should be a safe default for most users, particularly those who want to use `distcc`).

Please feel free to reopen if still not working for you. Best, sakaki

closed 2018-05-15 by sakaki-