sakaki-/gentoo-on-rpi-64bit #41: Lost DRI after update?

pronvit, 2018-03-26:
I installed the latest released image and was doing some OpenGL-related development. Installed some additional packages, and removed some that I didn't need to save space (mainly office, java, media players). Noticed that something was using CPU and eating disk space, it turned out to be the genup script (I thought it should start only after a week or so?). I waited for it to finish, then rebooted. After reboot, OpenGL didn't work anymore because DRI /dev/dri/* didn't exist, and X also failed to load software renderer as well (no such file). DRM and VC4 kernel modules were not loaded - not sure if they should be, if I load them manually nothing seems to happen. What went wrong?

sakaki-, 2018-03-26:
Apologies - I am working on a new release (with support for the RPi 3 B+) and enabled the 4.14 kernel; this may have been somewhat premature though, so I've now rolled back the mask. What kernel version do you have now (`uname -r`)?

pronvit, 2018-03-26:
Yep, 4.14.27-v8-4d78845fd711+ it is. Many years since I used Gentoo last time, so how do I downgrade now?

closed 2018-03-26 by pronvit

sakaki-, 2018-03-26:
Apologies, had to step away from my workstation there for a while: assume you got it sorted, since you closed the issue, but for reference (in case anyone else has the same problem), you just need to issue:
```console
pi64 ~ # emaint sync --repo rpi3
pi64 ~ # emerge -av1 bcmrpi3-kernel-bin
```

pronvit, 2018-03-26:
Yep, just a simple `emerge --sync && emerge bcmrpi3-kernel-bin` did downgrade.