sakaki-/gentoo-on-rpi-64bit #42: use "hdparm -z /dev/mmcblk0" instead of rebooting

tidux, 2018-03-27:
This forces a rescan of the partition table, and hdparm is known to work with mmcblk devices.

sakaki-, 2018-03-27:
Some people boot the image on non-sdcard devices though (there are a number of university users who have it set up this way); that's why the root is identified by UUID and the reboot is done. It only happens on first boot anyhow.

sakaki-, 2018-03-27:
Plus, and this is the more important point, afaik you cannot straightforwardly change the kernel's (active) partition table for a (non-vm, non-LVM) mounted root partition without a reboot, see e.g. [here](https://unix.stackexchange.com/questions/196435/is-it-possible-to-enlarge-the-partition-without-rebooting).

It _is_ possible to avoid a reboot if you e.g. switch root to a minimal ramfs, do the resize, reload the partition table, then switch root back, but it seemed simpler (given the one-time nature of the issue) just to reboot.

If this has changed (more recent kernel etc.) I'd be interested to know.

tidux, 2018-03-28:
Ah, I didn't realize it was all root.  In my usage of `hdparm -z` for live resize of disk images it's a separate storage volume on VMs.

closed 2018-04-08 by sakaki-