sakaki-/gentoo-on-rpi-64bit #43: Repos out of Date?

JaceAlvejetti, 2018-03-29:
Trying to install Glusterfs, Noticed emerge and eix only show 3.12 and 3.13 where https://packages.gentoo.org/packages/sys-cluster/glusterfs shows 4.0.0-r1,  3.13 was removed likely for an issue I fought with for quite a few hours (ipv6 only) 

So I ran genup then a manual emerge --sync

emerge complains about the repo being over a month old:

cat /usr/portage/metadata/timestamp.chk
Fri, 23 Feb 2018 00:45:01 +0000

I have forced the sync by killing the above file but it changes nothing.

Thanks,
Jace

JaceAlvejetti, 2018-03-29:
Found the change, /etc/portage/repos.conf/gentoo.conf swapped the comments around the sync-uri running emerge update currently. Is there a specific reason to run off the included repo? are you verifying packages?

Thanks, 
Jace

sakaki-, 2018-03-29:
Yes, I check that the packages in the isshoni.org repo build OK (those included on the image, that is) and they match the versions available (in binary form) in the binhost. Using that repo ensures that the two do not get out of sync. At the moment I am just preparing a new release (with support for the 3B+) so the packages have been frozen for a while, but the updates (to the binhost and repo) will come through in the next week or so.

JaceAlvejetti, 2018-03-29:
Thanks for the Update!

I will leave my production pi's on the official repo then, I have swapped one spare to the general gentoo repo, it is updating 148 packages with the help of the two other pi's and a gentoo x86 vm (12core) setup with your crossdev instructions. 

Looking forward to B+!

sakaki-, 2018-04-04:
v1.2.1 is out now (B+ supported); repo and binhost are unlocked again

closed 2018-04-04 by sakaki-