sakaki-/gentoo-on-rpi-64bit #44: Audio distortion from ALL application sources

aTosser, 2018-04-03:
I recently installed using the PINN installer and everything is working fantastic, with one exception.
All audio regardless of souce (VLC, firefox, Kodi, etc...) is distorted. Not to the point of misunderstanding the content but annoying none the less.

Is this related to [Sakaki's post here](https://forums.gentoo.org/viewtopic-p-8106974.html#8106974)?

aTosser, 2018-04-03:
Appologies, I intended to include HDMI in my comment. I have only a TV connected to the rPi3b.

sakaki-, 2018-04-03:
You could try changing some of the settings in your `/boot/config.txt` file, and rebooting, to see if they help (see e.g. [this listing](https://github.com/Evilpaul/RPi-config/blob/master/config.txt) for an explanation of the various keys).
For example, you could try appending:
```
config_hdmi_boost=7
```
which will increase the level of the HDMI signal (in case it is a long or poor quality HDMI cable causing the issue).
If that doesn't help, try
```
hdmi_force_edid_audio=1
```
(see above link for an explanation), and, if still no good, try appending:
```
hdmi_ignore_edid_audio=1
```
instead.

Does audio play back OK with the identical setup (cables, RPi3 board, TV) but using Raspbian?

aTosser, 2018-04-04:
Yes raspbian all works fine with no modifications to config.txt.  I would love to attempt the steps you suggested however I ran genup this morning now I am having boot issues. (Black screen) I don't know what to do to diagnose this so I will probably reinstall and try again.

sakaki-, 2018-04-04:
The v1.2.1 is out now (newer kernel) so perhaps that will help.

sakaki-, 2019-03-08:
Closing now - please re-open if this issue is still present with modern (>=1.4.0) releases of the image.
Best, sakaki

closed 2019-03-08 by sakaki-