sakaki-/gentoo-on-rpi-64bit #47: Refusing to create a cross-compiler using the same target name as your host utils.

IEXWER, 2018-05-14:
I don't get a gentoo linux pc,so i try to compile a linux, follow the wiki, on a raspberry 3b with img from you art work.
but I met this.

pi64 /usr/local/portage-crossdev/metadata # crossdev --stable -t aarch64-unknown-linux-gnu --genv 'USE="cxx multilib fortran -mudflap nls openmp -sanitize"'    
 * Refusing to create a cross-compiler using the same
 * target name as your host utils.

And i want to know how to fix it.
If i made some wrong please tell me.

IEXWER, 2018-05-14:
Ok,it seem to i have a aarch64-unknown-linux-gnu,

pi64 /usr/local/portage-crossdev/metadata # gcc-config -l
 [1] aarch64-unknown-linux-gnu-7.3.0

So i can't remake another.

sakaki-, 2018-05-14:
Hi, closing this, as you don't need a cross-compiler when building natively, as you have just noted.
Please also see my response to your related issue #48.
Best, sakaki

closed 2018-05-14 by sakaki-