sakaki-/gentoo-on-rpi-64bit #48: make: *** [Makefile:1021: vmlinux] Error 1 in raspberrypi-kernel_1.20180417-1

IEXWER, 2018-05-14:
OK, I still follow the wiki to compile a linux system in my raspberry 3b with img in this git.
When i compile kernel, the message show in my window.

lib/vsprintf.o:(___ksymtab+vsprintf+0x8): dangerous relocation: unsupported relo      cation
lib/vsprintf.o:(___ksymtab+vsscanf+0x0): dangerous relocation: unsupported reloc      ation
lib/vsprintf.o:(___ksymtab+vsscanf+0x8): dangerous relocation: unsupported reloc      ation
lib/vsprintf.o:(___ksymtab_gpl+bprintf+0x0): dangerous relocation: unsupported r      elocation
lib/vsprintf.o:(___ksymtab_gpl+bprintf+0x8): dangerous relocation: unsupported r      elocation
lib/vsprintf.o:(___ksymtab_gpl+bstr_printf+0x0): dangerous relocation: unsupport      ed relocation
lib/vsprintf.o:(___ksymtab_gpl+bstr_printf+0x8): dangerous relocation: unsupport      ed relocation
lib/vsprintf.o:(___ksymtab_gpl+vbin_printf+0x0): dangerous relocation: unsupport      ed relocation
lib/vsprintf.o:(___ksymtab_gpl+vbin_printf+0x8): dangerous relocation: unsupport      ed relocation
lib/win_minmax.o:(___ksymtab+minmax_running_max+0x0): dangerous relocation: unsu      pported relocation
lib/win_minmax.o:(___ksymtab+minmax_running_max+0x8): dangerous relocation: unsu      pported relocation
drivers/firmware/efi/libstub/lib-sort.stub.o: In function `__efistub_$d':
__efistub_sort.c:(.init___ksymtab+sort+0x0): dangerous relocation: unsupported r      elocation
__efistub_sort.c:(.init___ksymtab+sort+0x8): dangerous relocation: unsupported r      elocation
aarch64-unknown-linux-gnu-ld: warning: creating a DT_TEXTREL in a shared object.
make: *** [Makefile:1021: vmlinux] Error 1

I think, is there a wrong step in my operation?
My step is below.
Because the img has crossdev ,so i  don't need to setup crossdev.
Then i download the kernel source code from raspberry github.The kernel'branch is rpi-4.14.y.The release tag raspberrypi-kernel_1.20180417-1 Merge remote-tracking branch 'stable/linux-4.14.y' into rpi-4.14.y.

and type 
make ARCH=arm64 CROSS_COMPILE=aarch64-unknown-linux-gnu- distclean
make ARCH=arm64 CROSS_COMPILE=aarch64-unknown-linux-gnu- bcmrpi3_defconfig
make ARCH=arm64 CROSS_COMPILE=aarch64-unknown-linux-gnu- menuconfig
after this type, i just exit the menuconfig, delete .config file ,download .config file in this repository to replace it.
In the end, after type 
nice -n 19 make ARCH=arm64 CROSS_COMPILE=aarch64-unknown-linux-gnu- -j$(nproc)
the error emerge.The log message.
That's all what i have done.

My english is poor,If there has some place you can't understand,please tell me.
I will use the google translation to communicate with you.
Thank you.

IEXWER, 2018-05-15:
OK,I fix the problem and install my kernel.It works nice.Thanks.

But I want to know how to transplant another linux like ubuntu to this board.

sakaki-, 2018-05-15:
If you have another image or rootfs, even if it is 32bit, you can start by running it in a `chroot` from withi the booted `gentoo-on-rpi3-64bit` image, see e.g. my wiki post [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit/wiki/Running-32-bit-Raspbian-Packages-on-your-64-bit-Gentoo-System).

Otherwise, porting other distros is a bit out of scope for an issue on this project. I suggest you look/post on the "Operating system distributions" board of the Raspberry Pi forum ([here](https://www.raspberrypi.org/forums/viewforum.php?f=18)).

There is e.g. an Ubuntu 18.04 installer available from one of the contributors on there (see [this post](https://www.raspberrypi.org/forums/viewtopic.php?f=56&t=209763#p1309839)), but I haven't tried it myself. Good luck!

IEXWER, 2018-05-15:
Ok,thanks.I will try it.

sakaki-, 2018-05-15:
Closing as title issue has been addressed.

closed 2018-05-15 by sakaki-