sakaki-/gentoo-on-rpi-64bit #49: genup

snxq1995, 2018-05-16:
When I run genup
```
* Checking Portage configuration, please wait...
Usage: /usr/bin/emerge [-l <lexer> | -g] [-F <filter>[:<options>]] [-f <formatter>]
          [-O <options>] [-P <option=value>] [-s] [-v] [-x] [-o <outfile>] [<infile>]

       /usr/bin/emerge -S <style> -f <formatter> [-a <arg>] [-O <options>] [-P <option=value>]
       /usr/bin/emerge -L [<which> ...]
       /usr/bin/emerge -N <filename>
       /usr/bin/emerge -H <type> <name>
       /usr/bin/emerge -h | -V

Highlight the input file and write the result to <outfile>.

If no input file is given, use stdin, if -o is not given, use stdout.

If -s is passed, lexing will be done in "streaming" mode, reading and
highlighting one line at a time.  This will only work properly with
lexers that have no constructs spanning multiple lines!
```
I don't know that what happend.
By the way, This is my first time using gentoo.And It's awesome.

sakaki-, 2018-05-16:
Hi @snxq1995, sorry to hear you're having trouble with this ><

I have just run `genup` on a fresh version of the image to check, took quite a while to do the full update, but it ran to completion, and on reboot, `genup` still works.

On your system, it looks like `emerge --info` is failing. Are you running `genup` as root? Have you installed any other packages on your system?

What do you get if you run (as root):
```console
pi64 ~ # emerge --info
```

snxq1995, 2018-05-16:
use root  to get the same result.

You're right.this output  is the result of the `emerge --info`.But there is no first and last line.

I guess 
```
* Checking Portage configuration, please wait...
!@#!@#$balabala
* genup: Error: Caught signal - exiting
```

snxq1995, 2018-05-16:
use root  to get the same result.

You're right.this output  is the result of the `emerge --info`.But there is no first and last line.

Is the configuration problem?

OS Version: v1.2.1
```
* Checking Portage configuration, please wait...
!@#!@#$balabala
* genup: Error: Caught signal - exiting
```

snxq1995, 2018-05-16:
@sakaki-  Sorry,I'm a noob. Hope you don't mind.

snxq1995, 2018-05-16:
@sakaki- I saw your reply.Is there a relationship between emerge and python3.5?

Before this error occurs.I did this.
```
emerge --ask dev-lang/python:3.6
eselect python set 2(python3.6)
```

Haha
This is the reason?

sakaki-, 2018-05-16:
Not sure that'd be enough to cause it by itself, but you could try (as root) `eselect python set python3.5` and see if that resolves things (you should still have 3.5 installed, as it is in a different slot to 3.6).

If that _doesn't_ work, then please post the output I requested above (`head /usr/lib/python-exec/python3.5/emerge` etc.)

snxq1995, 2018-05-17:
emmm
I switched to Python3.5. But It does not work.
Here is the output of running `head /usr/lib/python-exec/python3.5/emerge`

```
pi64 /home/demouser/Projects # head /usr/lib/python-exec/python3.5/emerge 
#!/usr/bin/python3.5 -b
# Copyright 2006-2014 Gentoo Foundation
# Distributed under the terms of the GNU General Public License v2

from __future__ import print_function

import platform
import signal
import sys
```

sakaki-, 2018-05-17:
OK so please run (as root) `which emerge` and `ls -l $(which emerge)`.

snxq1995, 2018-05-17:
```
pi64 ~ # which emerge 
/usr/bin/emerge
pi64 ~ # ls -l $(which emerge)
lrwxrwxrwx 1 root root 31 Apr  1 15:39 /usr/bin/emerge -> ../lib/python-exec/python-exec2
```

sakaki-, 2018-05-17:
That's really very strange! Is there anything else you did (run `history` as root) or installed that may have changed your system? Its not something I have ever had reports of from other users.

Running `emerge` _should_ trigger `python-exec2` which should in turn invoke `/usr/lib/python-exec/python3.5/emerge` (which itself looks correct, based upon the `head` you posted), but instead it appears to be invoking `/usr/lib/python-exec/python3.5/pygmentize` (based upon the output you posted).

If there are no other obvious changes you made to your system, I would recommend posting in the "Gentoo on ARM" forum ([here](https://forums.gentoo.org/viewforum-f-62.html)), to see if anyone there can assist.

snxq1995, 2018-05-20:
OK.

I reinstall the image, but the problem is not reproduced.

Thank you.

closed 2018-05-20 by snxq1995

sakaki-, 2018-05-20:
Glad to hear you got it working.