sakaki-/gentoo-on-rpi-64bit #65: load gentoo-on-rpi3-64bit on Vmware or Qemo?

ardashir66, 2018-12-25:
Hi
Is it possible to load gentoo-on-rpi3-64bit image on Vmware or Qemo?
thanks in advance

sakaki-, 2018-12-29:
Hi ardashir66,

> Is it possible to load gentoo-on-rpi3-64bit image on Vmware or Qemo?

do you mean QEMU? It is certainly possible to `chroot` into the image on (e.g.) and x86_64 bit machine, and then run emerges etc. via **user**-mode QEMU via `binfmt_misc`: I have instructions for doing this on the project's wiki [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit/wiki/Build-aarch64-Packages-on-your-PC,-with-User-Mode-QEMU-and-binfmt_misc). Can be useful (given a sufficiently powerful machine) for emerging large packages etc. You can also (subject to init-system mismatches etc.) run some graphical apps from the `chroot`, and display them on your own system (via X11 forwarding).

As to **system** mode emulation, there is an issue with boot, since the RPi3's (currently unemulated) GPU controls the boot process. However, there is now some limited support for booting with the kernel directly using `-M raspi3`; see e.g. [this post](https://www.raspberrypi.org/forums/viewtopic.php?p=1386678#p1386678).

hth, sakaki

> PS I think this is possibly a duplicate of my [reply](https://www.raspberrypi.org/forums/viewtopic.php?p=1408714#p1408714) posted on the RPi3 forums (if you are user `arshad66` there).

ardashir66, 2019-01-01:
thank you very much for your answer

closed 2019-01-01 by ardashir66