sakaki-/gentoo-on-rpi-64bit #66: Pre-sync emerge in the image?

Avamander, 2019-01-18:
Would it be possible to pre-`--sync` emerge when creating the image? Other than that I love it, thanks!

sakaki-, 2019-01-19:
Thanks for the feedback ^-^

The image _is_ pre-sync'd as of the release date. But, the gated mirror of the main Gentoo repo (to which it is sync'd) is _locked_ for testing usually for 2 to 3 weeks prior to release, for stability.

After release, the mirror is unlocked, and the normal weekly binhost package build-bot is started up again (overall workflow is described [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit#new-features-to-expedite-regular-system-updating)). That means you are likely to get a largish update on first sync, as you 'catch up' with a month or so of changes to the tree (or more, of course, if you were to download an image released a month ago, or two months ago etc.)

closed 2019-01-20 by Avamander

Avamander, 2019-01-20:
Thank you for the answer! Your answer contained all I needed to know 👍 .