sakaki-/gentoo-on-rpi-64bit #67: Installing kernel modules like wireguard?

Avamander, 2019-01-20:
After using Gentoo a few days I'm really enjoying it, but there's basically no information how one could install and keep things like net-vpn/wireguard up-to-date on a custom kernel. Do I have to follow the [Optional: Compiling a kernel from source](https://github.com/sakaki-/gentoo-on-rpi3-64bit#optional-compiling-a-kernel-from-source) guide or is there a nicer way to generate the necessary configuration `sudo emerge --ask net-vpn/wireguard` requires?

I'm currently getting this error:
```
 * Could not find a usable .config in the kernel source directory.
 * Please ensure that /usr/src/linux points to a configured set of Linux sources.
 * If you are using KBUILD_OUTPUT, please set the environment var so that
 * it points to the necessary object directory so that it might find .config.
 * ERROR: net-vpn/wireguard-0.0.20181218::gentoo failed (setup phase):
 *   Kernel not configured; no .config found in /usr/src/linux
```

sakaki-, 2019-01-20:
OK, there are two issues here: the [required configuration](https://www.wireguard.com/install/#kernel-requirements) of the host kernel itself (to support wireguard), and building the wireguard module out-of-tree for insertion into it.

Issue 1
---------

So the image uses a binary kernel package which uses [this weekly bot-built kernel](https://github.com/sakaki-/bcmrpi3-kernel-bis) as its target.

This kernel is based upon the standard upstream `bcmrpi3_defconfig`, but has a number of additional config features enabled. You can submit a PR to add features to the [`conform_config.sh` script](https://github.com/sakaki-/bcmrpi3-kernel-bis/blob/master/conform_config.sh), and if merged the next weekly autobuild will include your changes. At each weekly build point, a matching ebuild is automatically created [here](https://github.com/sakaki-/rpi3-overlay/tree/master/sys-kernel/bcmrpi3-kernel-bis-bin). If you unmask a particular version (add an entry to `/etc/portage/package.mask` - you can see the profile default versions [here](https://github.com/sakaki-/rpi3-overlay/blob/master/profiles/targets/rpi3/package.unmask/bcmrpi3-kernel-bis-bin)), then the new binary kernel will be installed on your next `@world` update (or `genup`). You can use this to add any missing kernel requirements for wireguard, and once done they will be in the autobuild going forward. I think most/all of the requirements for wg should already be in there (check against `/proc/config.gz` (may have to `modprobe configs` first) or `/boot/config`), so this issue may be moot.

Issue 2
---------
However, the other issue is that wireguard is actually built (normally, and by the gentoo `ebuild`) as an _out-of-tree_ kernel module. So, it needs appropriate sources available in `/usr/src/linux`. One way around this is to build a kernel manually (using the 'live' config from `/proc/config.gz`) (you already have the link above ^-^), _or_ you could clone the appropriate source tree into `/usr/src/` and link `/usr/src/linux` to it.

For the latter approach, if you use `uname -r` to get the kernel release name, the end of this  will include the short-form tip commit at the time of build. Then, you can simply clone the appropriate tree at that commit from [here](https://github.com/raspberrypi/linux).

I might look at creating a matching 'kernel headers' package as part of the autobuild in future, so you could [build against those](https://www.cyberciti.biz/tips/build-linux-kernel-module-against-installed-kernel-source-tree.html) instead of having to pull the full tree. But that isn't an option right now, sorry about that ><

Avamander, 2019-01-21:
After following the guide in the README I had to at least check "[*] Parallel crypto engine", I'm not sure if there was anything else because I also previously checked everything needed for nftables. I'll see how far I can get with the custom kernel...

Avamander, 2019-01-21:
Small question though, there's a bunch of arm64 cryptographic acceleration options, I assume the NEON and just ARMv8 extension (those not requiring ARM's cryptographic core) ones can be safely enabled?

sakaki-, 2019-01-22:
Yes, you should be fine with those. If you get a working config with your custom kernel for wg, then please consider submitting a PR for bcmrpi3-kernel-bis. Thanks!

sakaki-, 2019-08-17:
Old issue, closing now in pre 1.5.0 housecleaning

closed 2019-08-17 by sakaki-