sakaki-/gentoo-on-rpi-64bit #68: Permission Denied

tdubz1, 2019-01-21:
I have download Pocketmine-MP on the image you developed for the RPi3b+ - Thank you very much btw -works v/ well however when I try to edit the php.ini file to enable an extention for a Pocketmine Plugin it won't let me edit it and comes up with the message 'Permission Denied' when I try and save the changes.

Thanks, tdubz

sakaki-, 2019-01-21:
Hi @tdubz1,

the file `php.ini` is writeable only by the root user. To get around this, open a terminal and issue:

```console
demouser@pi64 ~ $ sudo mousepad &>/dev/null&
```

This will open the graphical editor `mousepad` with elevated privileges, that can be used to edit, and save, the `php.ini` file. Don't use it for editing normal files!

hth, sakaki

PS if the image has worked for you, perhaps you could mark your original post as 'solved' on the RPi forums, for the benefit of anyone else who comes across your question by google etc. Thanks!

tdubz1, 2019-01-21:
Thank you! I have marked my post as closed.

sakaki-, 2019-01-22:
Closing this issue now. Please re-open if the `sudo mousepad` didn't work for you.
Best, sakaki

closed 2019-01-22 by sakaki-