sakaki-/gentoo-on-rpi-64bit #69: Kodi -> Gear menu -> System crashes

Avamander, 2019-01-21:
I just wanted to check out what Kodi looks like but for some reason it crashes when I launch it, press the Gear icon and from there "System" option. If I should report this in Kodi repository then please do let me know and I'll do so. 

The error that gets printed to terminal is:
```
terminate called after throwing an instance of 'fmt::v5:format_error'
  what(): invalid type specifier
/usr/bin/kodi: line 219: 4064 Aborted ${KODI_BINARY} $SAVED_ARGS
Crash report available at /home/demouser/kodi_crashlog-[crash date and time].log
```

The crash log does not contain a stacktrace because gdb is not installed, if it's required I can do so.

sakaki-, 2019-01-21:
I'm away from my workstation right now, so can't check this - could you try _downgrading_ kodi and see if the error persists? These will be binary package installs so hopefully not time-consuming. Try:

```console
demouser@pi64 ~ $ sudo emerge -av1 =media-tv/kodi-18.0_rc4
```

to drop back from rc5 to rc4, and if the problem is still there, then:

```console
demouser@pi64 ~ $ sudo emerge -av1 =media-tv/kodi-17.4_rc1
```

to go back a major version. Once done,

```console
demouser@pi64 ~ $ sudo emerge -av1u media-tv/kodi
```
will get you back up to date again.

Can you play media files (movies, audio) from kodi OK?

Thanks!

sakaki-, 2019-02-03:
Kodi has [just been bumped](https://isshoni.org/pi64pie/media-tv/?C=M;O=D) to 18.0 (non-rc) and the "gear" -> "system" operation seems to be working now. Just run `sudo genup` to get it (or wait for the weekly autoupdate to do this for you).

Closing now (as a fixed upstream issue I think) but please re-open if still not functional once you get the upgraded version.

Thanks for reporting this!

Best, sakaki

closed 2019-02-03 by sakaki-

Avamander, 2019-02-04:
Thank you for testing this, I've been a bit busy. I'll try updating and check how it works.