sakaki-/gentoo-on-rpi-64bit #70: Tabs

tdubz1, 2019-02-02:
Not really an issue but for some reason my tabs have moved to the right of the screen instead of the left. How do I fix this? Thanks TDubz

tdubz1, 2019-02-02:
Desktop

sakaki-, 2019-02-02:
Perhaps you have moved the transparent expandable separator in the panel layout (or added a second separator)? Left-click on the top panel (in a gray area with no button or other icon in it) and select <kbd>Panel</kbd>&rarr;<kbd>Panel Preferences...</kbd>. Then click on the <kbd>Items</kbd> tab. Where do you see the "Separator" item in the list?

tdubz1, 2019-02-03:
Thank you, that has helped! It is now back to normal! (The seperator was at the top!)

sakaki-, 2019-02-03:
Hi @tdubz1 - glad that helped ^-^
However, I see you've closed and then reopened the issue; are you still experiencing problems with this? Best, S.

tdubz1, 2019-02-04:
No, sorry I didn’t mean to open the issue again. I will close the issue now. Thanks for your help.

closed 2019-02-04 by tdubz1