sakaki-/gentoo-on-rpi-64bit #72: Can't logon

tdubz1, 2019-02-12:
When I booted up my Raspberry Pi 3B+ running your Gentoo Linux  It came to page saying I am in 'none'! I don't logon to the regular login screen and am met with terminal like thing that I can't get out of,

Please help me,

tdubz

sakaki-, 2019-02-12:
Can you log in as demouser at the terminal prompt? Looks like your root filesystem may have become corrupted somehow; did your system suffer a hard powerdown or similar?

tdubz1, 2019-02-13:
No, I can’t login as demouser because I deleted that account and made a new one but I can login to that one. Yes, I think it did suffer a hard powerdown (when I turned the power off to the Pi).

sakaki-, 2019-02-13:
Can you log in as your new user at the terminal prompt? Or as root (the default password for root is raspberrypi64, unless you changed that too)?

tdubz1, 2019-02-13:
Yes I can

sakaki-, 2019-02-13:
OK, so logon as root, and run `findmnt`; what output do you get?

tdubz1, 2019-02-13:
https://drive.google.com/file/d/1T7VRedXmiwIWpM47CCRecSw3sz9H-VZc/view?usp=sharing

sakaki-, 2019-02-13:
OK so as expected your root partition is read only (first line of output, the `ro` tag).
Try running (as root)
```console
# e2fsck -f /dev/mmcblk0p2
```
Once it completes, try rebooting.
If `e2fsck` reports errors, you should **backup the card** before proceeding, and then reboot, run the command again, and answer y when prompted. Then try rebooting again, with luck your system will come up.

tdubz1, 2019-02-13:
Thank you! I am now able to log in! 
I am very thankful and appreciate your time.
tdubz

sakaki-, 2019-02-13:
OK - good news!
It's quite a common thing to happen on Linux, if your system suffers a hard poweroff rather than a clean shutdown.
I'll close this issue now.
Best,
sakaki

closed 2019-02-13 by sakaki-