sakaki-/gentoo-on-rpi-64bit #73: Wi-Fi router - configuration hostapd

JBielanski, 2019-02-14:
I tried to configure WiFi access point and my every attempts have failed.
I used following tutorials:
https://wiki.gentoo.org/wiki/Home_router
https://wiki.gentoo.org/wiki/Hostapd
When I try to start hostapd service then during starting net.wlan0 it's looking for connection to 
network instead of using hostapd configuration.

sakaki-, 2019-02-14:
Are you still using NetworkManager? If so, you probably either need to disable it, or to add wlan0 to its list of ignored interfaces (see e.g. [here](https://askubuntu.com/questions/21914/how-can-i-make-networkmanager-ignore-my-wireless-card)).

sakaki-, 2019-08-07:
@JBielanski,
I believe you have posted the solution yourself to this problem on the Gentoo forums [here](https://forums.gentoo.org/viewtopic-p-8355076.html#8355076)?
Your answer copied below for others who may come to this by search engine:

> Problem has been solved. The source of problem was configuration file: 

```ini
# DNS Domain
dns_domain_lo="astro-pi64.org"

# CONFIG eth0 [network access]
config_eth0="dhcp"

# WIFI ACCESS POINT
# ** network connection
#modules_wlan0="wpa_supplicant"
#config_wlan0="dhcp"
# ** access point
modules_wlan0="[b]!iw[/b] !iwconfig !wpa_supplicant"
config_wlan0="192.168.66.3/24 brd 192.168.66.0"
routes_wlan0="default via 192.168.66.1"
rc_net_wlan0_provide="!net" 
```
 
> I missed to exclude completely iw interface (!iwconfig is not enough, I have to exclude also !iw). I had seen this mistake when I tried to setup AP on on computer for testing.

Thanks for posting this, closing now. Please re-open if I have got the wrong end of the stick, and these issues are not related.

closed 2019-08-07 by sakaki-