sakaki-/gentoo-on-rpi-64bit #76: "Temporary failure in name resolution"

tdubz1, 2019-02-21:
Whenever I try and wget content or ping an IP/website I get the error "Temporary failure in name resolution" and when I try and load a website the web-page never loads however weirdly I can ping 8.8.8.8.

Please help me,

tdubz

sakaki-, 2019-02-21:
What do you get if you run `cat /etc/resolv.conf`?

tdubz1, 2019-02-21:
I get:

`# Generated by NetworkManager`
`nameserver 208.67.220.220`
`nameserver 194.72.9.34`
`nameserver 8.8.8.8`

sakaki-, 2019-02-21:
It may be that some of the earlier (i.e., higher priority) servers in your DNS list are failing or rejecting your requests occasionally.

You could try setting `8.8.8.8` as your _primary_ DNS. To do this, _right_-click on the network icon in the panel (top right of your screen) and select <kbd>Edit Connections...</kbd> from the context menu. In the dialog that then opens, select your current network connection and (left) click on the 'gearwheel' icon. Then choose the <kbd>IPv4 Settings</kbd> tab, and in the `Method` drop-down, choose <kbd>Automatic (DHCP) addresses only</kbd>. Then, in the `DNS servers` field, enter `8.8.8.8,208.67.220.220,194.72.9.34` and then press <kbd>Save</kbd>. Now _disconnect_ from your current network connection, wait a minute or so, then reconnect again. Then issue `cat /etc/resolv.conf` again. You should now see that Google's nameserver, `8.8.8.8` is at the _top_ of the list.

Does this make things any more reliable?

tdubz1, 2019-03-05:
No, it didn't. For some reason when I reset the internet on my Pi, the internet is temporarily accessible for around 24 hrs however when this time is over it displays the error again.

For some reason when I execute this command now I get:
`# Generated by NetworkManager
nameserver: 192.168.17.134`

P.S. Sorry I haven't answered this sooner.

sakaki-, 2019-03-05:
But now, when you are getting the error, what does `/etc/resolv.conf` contain? Is `8.8.8.8` at the top of the list now?

tdubz1, 2019-03-09:
Closing issue now because I have found solution.

closed 2019-03-09 by tdubz1

sakaki-, 2019-03-09:
@tdubz1,
glad to hear you go this resolved! Is there anything you did to solve the issue that I should include in the project's README?
Best, sakaki

fifn2, 2019-06-20:
@tdubz1, I'm having the same problem. Would you post your solution, please?

sakaki-, 2019-06-20:
First, try setting `8.8.8.8` as your primary DNS, as suggested [above](https://github.com/sakaki-/gentoo-on-rpi3-64bit/issues/76#issuecomment-466193092).

fifn2, 2019-06-20:
@sakaki- Hmm. That doesn't seem to have worked -- unless those IP addresses are the exact ones you wanted me to type in.

sakaki-, 2019-06-20:
@fifn2, are you connected over WiFi or Ethernet? Can you please post the contents of `cat /etc/resolv.conf`? Also, if you try `ping -c 3 8.8.8.8`, what response do you get?

fifn2, 2019-06-20:
I'm connected over wifi.
`/etc/resolv.conf`:
```
# Generated by NetworkManager
nameserver 8.8.8.8
nameserver 192.168.1.11
```
`ping -c 3 8.8.8.8`:
```
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
From 192.168.1.15 icmp_seq=1 Destination Host Unreachable
From 192.168.1.15 icmp_seq=2 Destination Host Unreachable
From 192.168.1.15 icmp_seq=3 Destination Host Unreachable

--- 8.8.8.8 ping statistics ---
3 packets transmitted, 0 received, +3 errors, 100% packet loss, time 2038ms
pipe 3
```

sakaki-, 2019-06-20:
@fifn2,

OK, these results indicate a problem with the WiFi connection itself, not a DNS issue per se (since you cannot ping a numeric IP address, which does not require name resolution).

However, your `/etc/resolv.conf` output suggests (unless you have set something manually beforehand) that (?) your router is on `192.168.1.11`, and this IP address has been provided as the default DNS. That suggests your system made at least the initial connection over WiFi successfully.

Were you able to browse the net at all before the name resolution error showed up? Also, please post the output of `ping -c 3 192.168.1.11`; this will show if you are able to send packets to and from the router. Also please post the result of `ifconfig`.

Thanks, sakaki

fifn2, 2019-06-21:
@sakaki-,
I don't think I was able to make the initial WiFi connection successfully. It also might be worth noting that after I tried to configure the `/etc/resolv.conf` file with the GUI, I modified the file manually. I'm willing to do a fresh install if you think it's necessary.
`ping -c 3 192.168.1.11`:
```
PING 192.168.1.11 (192.168.1.11) 56(84) bytes of data.

--- 192.168.1.11 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2028ms
```
`ifconfig`:
[image]

sakaki-, 2019-06-21:
From the `ifconfig` output, it seems at least some packets have been
transferred in each direction. Are you running the latest (1.4.2) image?

Does your RPi3 work OK connected to the same wifi router when running
stock Raspbian?

If so, it'd be very useful if you could try downloading and running my
raspbian-nspawn-64 image
(https://github.com/sakaki-/raspbian-nspawn-64), as this will help show
whether then 64-bit kernel or Gentoo userspace is at issue here (as the
raspbian-nspawn-64 image shares the same kernel with the gentoo one, but
has a stock Raspbian userland).

Another useful thing would be to post the output of `dmesg | grep -i
wlan` (on your current Gentoo system); if the connection is being
dropped or is having negotiation issues, this should show it up.

Best, sakaki

fifn2, 2019-06-21:
RPi3(B+) works fine connected to the same wifi router on raspbian.
`dmesg | grep -i wlan` on Gentoo:
```
[    41.355367] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    41.400295] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    41.916122] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
```

sakaki-, 2019-06-21:
@fifn2,

what WiFi [regulatory domain](https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2#Decoding_table) do you use in Raspbian? The Gentoo system ships with the `GB` regulatory domain set (in `/etc/conf.d/rpi3-wifi-regdom`). If you are in a different region, try changing this file, to contain the appropriate code for your location, and restarting.

Also, are you using the regular or 5GHz WiFi bands?

sakaki-, 2019-06-21:
@fifn2,

one other quick test that would be useful: on Gentoo, what happens if you _disconnect_ your WiFi connection (using the NetworkManager menu, top right of your screen) and then _reconnect_ to your chosen access point again (via the same menu). Do the ping etc. tests come out the same after reconnection?

fifn2, 2019-06-22:
@sakaki- 
I had already set the file... but now the wifi's working (!). I have no idea why; I think sometimes wifi just doesn't like me 😅 .

sakaki-, 2019-06-22:
@fifn2,
glad to hear it is working now ^-^
sakaki