sakaki-/gentoo-on-rpi-64bit #78: How do I enable the sqlite3 extension for my Pocketmine Server?

tdubz1, 2019-03-09:
How do I enable the sqlite3 extension for my Pocketmine Server?

sakaki-, 2019-03-13:
Could you please post the error message you get from `pocketmine`?
Thanks, sakaki

tdubz1, 2019-03-14:
[https://drive.google.com/file/d/1S2_Iiv4_mAciD0sLieRSaJ3CawanmhSO/view](url)

sakaki-, 2019-03-14:
You need to re-emerge the `dev-lang/php` package with the `sqlite` USE flag set. To do do, issue
```console
demouser@pi64 ~ $  echo "dev-lang/php sqlite" | sudo tee -a /etc/portage/package.use/php
```

then:
```console
demouser@pi64 ~ $  nice -n 19 sudo emerge --verbose --oneshot dev-lang/php
```

This may take some time to complete, so leave it to run overnight perhaps. Once done, if you restart your pocketmine server, you should (hopefully) find you can load the sqlite3 extension.

tdubz1, 2019-03-15:
Thank you very much @sakaki- I am very thankful for your help,
The plugin I want to run is now running smoothly!

closed 2019-03-15 by tdubz1