sakaki-/gentoo-on-rpi-64bit #80: Enable MySQli extension?

tdubz1, 2019-03-24:
Same issue as the sqlite3 extension as I opened up earlier. Whenever I remove the semi-colon in the php.ini the plugin (requiring MySQli) comes up with this error: [https://drive.google.com/open?id=1BDJ4Q0F-7Pim5rB5jsy4k7uXfl0uBoiX](url)

Thanks,

tdubz

sakaki-, 2019-03-27:
You need to add the `mysqli` USE flag to `dev-lang/php`, and re-emerge it.

For convenience, I have [added this flag](https://github.com/sakaki-/rpi3-overlay/commit/f13697be3dba1688c1f0128ec99c261411590b20) (and `sqlite`) to the binary package build on the isshoni.org server, rebuilt it and pushed the resulting package to the binhost.

To get it, issue:
```console
demouser@pi64 ~ $ sudo emaint sync --auto
demouser@pi64 ~ $ sudo emerge -av dev-lang/php
```
Hopefully this will automatically use the binary package just mentioned, rather than compiling from scratch locally, so should be relatively quick.

After that, if you restart your pocketmine server, with luck you will be able to enable the desired extension.

sakaki-, 2019-04-25:
Closing now. Please re-open if the above does not work for you.

closed 2019-04-25 by sakaki-