sakaki-/gentoo-on-rpi-64bit #82: Gtkwebkit and qtwebkit binaries

l0ner, 2019-03-28:
Hi

It would be nice to have dev-qt/qtwebkit and gtkwebkit on the binhost, as they are absolute behemoths to compile (I gave up trying to emerge gtkwebkit after 15 days of compilation), since the "lightweight" web-browsers like midori or otter depend on them.

sakaki-, 2019-03-28:
OK, so I've added `dev-qt/qtwebkit` and `net-libs/webkit-gtk` (I assume that's what you meant by "gtkwebkit"?) to the server autobuild list, built them (the server uses a [QEMU user-mode chroot](https://github.com/sakaki-/gentoo-on-rpi3-64bit/wiki/Build-aarch64-Packages-on-your-PC,-with-User-Mode-QEMU-and-binfmt_misc) for efficiency), and pushed the resulting binary packages to the [binhost](https://isshoni.org/pi64pie/) (the current versions are `dev-qt/qtwebkit-5.212.0_pre20180120` and `net-libs/webkit-gtk-2.24.0`).

You should be able to get these, via binary package, if you do:
```console
demouser@pi64 ~ $ sudo emaint sync --auto
demouser@pi64 ~ $ sudo emerge -av dev-qt/qtwebkit net-libs/webkit-gtk
```

Please let me know if they work for you!

Best, sakaki

l0ner, 2019-04-02:
Thanks a lot, having those two in binhost did help a lot in emerging midori and otter.

Another package that seem to take long time to compile is dev-qt/qtwebengine, required by www-client/falkon, it would be great if you could add that to binhost too.

sakaki-, 2019-04-02:
Sure, I'll have a look at the end of this week.
Best, S.

sakaki-, 2019-04-05:
I've just pushed `dev-qt/qtwebengine-5.12.2` to the binhost (you're right - it was a bit of a beast to build ^-^)
It's on the autobuild list now also.
hth,
sakaki

l0ner, 2019-04-07:
Thanks a lot! I'll sync my portage and install it as soon as I get home.

sakaki-, 2019-04-25:
Closing now. Please re-open if there are any issues.

closed 2019-04-25 by sakaki-

l0ner, 2019-06-01:
Hi, it's beed a while since I had time to look at this.

Anyway i have two issues:

1. `www-client/falkon` requires `dev-qt/qtwebengine` with `widgets` use flag enabled:

```
~ > emerge -pv falkon

Local copy of remote index is up-to-date and will be used.

These are the packages that would be merged, in order:

Calculating dependencies                    ... done!           
[binary  N     ] app-arch/snappy-1.1.7:0/1::gentoo  USE="(-static-libs) -test" 0 KiB
[binary  N     ] dev-libs/re2-0.2018.10.01:0/gentoo-2017-03-01::gentoo  USE="-icu" 0 KiB
[binary  N     ] media-libs/libvpx-1.8.0-r1:0/6::gentoo  USE="highbitdepth postproc svc threads -doc -static-libs -test" 0 KiB
[binary  N     ] media-libs/x264-0.0.20190214:0/157::gentoo  USE="interlaced threads (-altivec) (-opencl) -pic -static-libs" 0 KiB
[binary  N     ] media-libs/xvid-1.3.5::gentoo  USE="threads -examples -pic" 0 KiB
[binary  N     ] media-libs/libtheora-1.1.1-r2::gentoo  USE="encode -doc -examples -static-libs" 0 KiB
[binary  N     ] media-sound/lame-3.100-r1::gentoo  USE="frontend -debug -mp3rtp -sndfile -static-libs" 0 KiB
[binary  N     ] media-libs/libv4l-1.16.3::gentoo  USE="jpeg" 0 KiB
[ebuild     U  ] kde-frameworks/extra-cmake-modules-5.58.0:5/5.58::gentoo [5.57.0:5/5.57::gentoo] USE="-doc -test" 319 KiB
[binary  N     ] media-video/ffmpeg-4.1.3:0/56.58.58::gentoo  USE="X alsa bzip2 chromium encode gpl hardcoded-tables iconv libv4l mp3 network opengl openssl postproc sdl svg theora threads truetype v4l vorbis x264 xcb xvid zlib (-altivec) (-amr) -amrenc (-appkit) (-bluray) (-bs2b) -cdio -chromaprint -codec2 -cpudetection -debug -doc -fdk -flite -fontconfig -frei0r -fribidi -gcrypt -gme -gmp -gnutls -gsm -iec61883 (-ieee1394) -jack -jpeg2k -kvazaar (-ladspa) -libaom -libass -libcaca -libdrm -libilbc -libressl -librtmp -libsoxr -libxml2 -lv2 -lzma (-mipsdspr1) (-mipsdspr2) (-mipsfpu) (-mmal) -modplug -openal (-opencl) -openh264 -opus -oss -pic -pulseaudio -rubberband -samba -snappy -speex -srt -ssh -static-libs -test -twolame -vaapi (-vdpau) (-vpx) -wavpack -webp -x265 -zeromq -zimg (-zvbi)" CPU_FLAGS_ARM="(neon) (thumb) (thumb2) (v6) (v8) (vfp) (vfpv3)" FFTOOLS="aviocat cws2fws ffescape ffeval ffhash fourcc2pixfmt graph2dot ismindex pktdumper qt-faststart sidxindex trasher" VIDEO_CARDS="(-nvidia)" 9,933 KiB                        
[binary  N     ] dev-qt/qtwebchannel-5.12.3:5/5.12::gentoo  USE="qml -debug -test" 95 KiB
[ebuild  N     ] kde-frameworks/ki18n-5.58.0:5/5.58::gentoo  USE="-debug -doc -test" PYTHON_SINGLE_TARGET="python3_6 -python2_7 -python3_5 (-python3_7)" PYTHON_TARGETS="python2_7 python3_5 python3_6 (-python3_7)" 575 KiB                                                                                                                                                                                                      
[ebuild  N     ] dev-qt/qtwebengine-5.12.3:5/5.12::gentoo  USE="alsa bindist system-ffmpeg system-icu widgets -debug -designer -geolocation -jumbo-build -pax_kernel -pulseaudio -test" 243,434 KiB
[ebuild  N     ] www-client/falkon-3.1.0-r1::gentoo  USE="X dbus -debug -kde -libressl -test" 2,239 KiB

Total: 14 packages (1 upgrade, 13 new, 10 binaries), Size of downloads: 256,592 KiB

The following USE changes are necessary to proceed:
 (see "package.use" in the portage(5) man page for more details)
# required by www-client/falkon-3.1.0-r1::gentoo
# required by falkon (argument)
>=dev-qt/qtwebengine-5.12.3 widgets

!!! The following installed packages are masked:
- app-text/mupdf-1.14.0-r3::gentoo (masked by: package.mask)
/usr/local/portage/rpi3/profiles/targets/rpi3/package.mask/mupdf:
# build errors on arm64
# mask to prevent pull-in of non-bindist openssl (for now)

For more information, see the MASKED PACKAGES section in the emerge
man page or refer to the Gentoo Handbook.

emerge: there are no ebuilds built with USE flags to satisfy ">=dev-qt/qtwebengine-5.11.1:5=[widgets]".
!!! One of the following packages is required to complete your request:
- dev-qt/qtwebengine-5.11.3::gentoo (Change USE: +widgets)
(dependency required by "www-client/falkon-3.1.0-r1::gentoo" [ebuild])
(dependency required by "falkon" [argument])
````

Could you please enable it on binhost, as www-client/falkon is the only reason i'd want the dev-qt/qtwebengine on my system?

2. I don't know why but emerge insists on building `dev-qt/qtwebengine` from source, instead of pulling it from binhost (i tried to pull it by itself with default use flags, and emerge still wants to build it). I did `emaint sync --auto`, i did `eix-sync`, i did `emerge --sync` and no dice. Any ideas?

sakaki-, 2019-06-02:
I'll add that flag and rebuild sometime this coming week.
As to the build-from-source issue, it may be that the dependencies of the package have changed. You can try emerging it with `--binpkg-changed-deps=n --binpkg-respect-use=n` and the binary package should be picked up (but I don't recommend using these flags to `emerge` for day-to-day use, for obvious reasons!).

l0ner, 2019-06-05:
i have poked a little, tried what you've recommended and emerge still insists on building =dev-qt/qtwebengine-5.12.3 from source. Strange thing is it has no problems with installing 5.12.2 from binary.

Emerging 5.12.3:
```
 > emerge -pvt dev-qt/qtwebengine

Local copy of remote index is up-to-date and will be used.

These are the packages that would be merged, in reverse order:

Calculating dependencies             ... done! 
[ebuild  N     ] dev-qt/qtwebengine-5.12.3:5/5.12::gentoo  USE="alsa bindist system-ffmpeg system-icu -debug -designer -geolocation -jumbo-build -pax_kernel -pulseaudio -test -widgets" 243,434 KiB
[binary  N     ]  media-libs/libvpx-1.8.0-r1:0/6::gentoo  USE="highbitdepth postproc svc threads -doc -static-libs -test" 0 KiB
[binary  N     ]  dev-libs/re2-0.2018.10.01:0/gentoo-2017-03-01::gentoo  USE="-icu" 0 KiB
[binary  N     ]  dev-qt/qtwebchannel-5.12.3:5/5.12::gentoo  USE="qml -debug -test" 95 KiB
[binary  N     ]  app-arch/snappy-1.1.7:0/1::gentoo  USE="(-static-libs) -test" 0 KiB

Total: 5 packages (5 new, 4 binaries), Size of downloads: 243,528 KiB
```

Emerging 5.12.2 (downgrades are caused by qt requiring all of it's components to be the same version, and I have already installed some stuff from qt-5.12.3)
```
portage > emerge -pvt =dev-qt/qtwebengine-5.12.2

Local copy of remote index is up-to-date and will be used.

These are the packages that would be merged, in reverse order:

Calculating dependencies            ... done!                         
[binary  N     ] dev-qt/qtwebengine-5.12.2:5/5.12::gentoo  USE="alsa bindist system-ffmpeg system-icu -debug -designer -geolocation -jumbo-build -pax_kernel -pulseaudio -test -widgets" 48,585 KiB
[binary  N     ]  dev-qt/qtwebchannel-5.12.2:5/5.12::gentoo  USE="qml -debug -test" 95 KiB
[binary     UD ]  dev-qt/qtprintsupport-5.12.2:5/5.12::gentoo [5.12.3:5/5.12::gentoo] USE="cups -debug (-gles2) -test" 0 KiB
[binary     UD ]  dev-qt/qtdeclarative-5.12.2:5/5.12::gentoo [5.12.3:5/5.12::gentoo] USE="widgets -debug (-gles2) (-jit) -localstorage -test" 0 KiB
[binary     UD ]   dev-qt/qtwidgets-5.12.2:5/5.12::gentoo [5.12.3:5/5.12::gentoo] USE="gtk png xcb -debug (-gles2) -test" 0 KiB
[binary     UD ]    dev-qt/qtgui-5.12.2:5/5.12::gentoo [5.12.3-r1:5/5.12::gentoo] USE="dbus egl gif jpeg libinput png udev xcb -accessibility -debug -eglfs -evdev (-gles2) -ibus -test -tslib -tuio -vnc" 0 KiB
[binary     UD ]   dev-qt/qttest-5.12.2:5/5.12::gentoo [5.12.3:5/5.12::gentoo] USE="-debug -test" 0 KiB
[binary     UD ]   dev-qt/qtnetwork-5.12.2:5/5.12::gentoo [5.12.3:5/5.12::gentoo] USE="bindist ssl -connman -debug (-libproxy) -networkmanager -sctp -test" 0 KiB
[nomerge       ] dev-qt/qtwebengine-5.12.2:5/5.12::gentoo  USE="alsa bindist system-ffmpeg system-icu -debug -designer -geolocation -jumbo-build -pax_kernel -pulseaudio -test -widgets" 
[nomerge       ]  dev-qt/qtgui-5.12.2:5/5.12::gentoo [5.12.3-r1:5/5.12::gentoo] USE="dbus egl gif jpeg libinput png udev xcb -accessibility -debug -eglfs -evdev (-gles2) -ibus -test -tslib -tuio -vnc" 
[binary     UD ]   dev-qt/qtdbus-5.12.2:5/5.12::gentoo [5.12.3:5/5.12::gentoo] USE="-debug -test" 0 KiB
[binary     UD ]    dev-qt/qtcore-5.12.2:5/5.12::gentoo [5.12.3:5/5.12::gentoo] USE="icu -debug -systemd -test" 0 KiB
[binary  N     ]  media-libs/libvpx-1.7.0:0/5::gentoo  USE="highbitdepth postproc svc threads -doc -static-libs -test" 1,068 KiB
[binary  N     ]  dev-libs/re2-0.2018.10.01:0/gentoo-2017-03-01::gentoo  USE="-icu" 0 KiB
[binary  N     ]  app-arch/snappy-1.1.7:0/1::gentoo  USE="(-static-libs) -test" 0 KiB
[nomerge       ] dev-qt/qtwebengine-5.12.2:5/5.12::gentoo  USE="alsa bindist system-ffmpeg system-icu -debug -designer -geolocation -jumbo-build -pax_kernel -pulseaudio -test -widgets" 
[nomerge       ]  dev-qt/qtcore-5.12.2:5/5.12::gentoo [5.12.3:5/5.12::gentoo] USE="icu -debug -systemd -test" 
[binary     UD ]   dev-libs/double-conversion-3.0.3:0/2::gentoo [3.1.4-r1:0/3::gentoo] USE="-static-libs -test" 67 KiB

Total: 14 packages (9 downgrades, 5 new, 14 binaries), Size of downloads: 49,813 KiB
```

On another note: do we have dev-qt/qtwebchannel on binhost? Do we have it with qml flag enabled, I guess yes, but asking never hurts.

dony71, 2020-03-12:
I'm trying to save some space
Can you tell me how to uninstall this qtwebkit ?
They are not in /var/lib/portage/world