sakaki-/gentoo-on-rpi-64bit #85: Support for BTRFS root

Necktwi, 2019-05-13:
Please compile in `BTRFS` into the kernel.

sakaki-, 2019-05-15:
If you submit an appropriate PR for [`bcmrpi3-kernel-bis`](https://github.com/sakaki-/bcmrpi3-kernel-bis) (see e.g. [this example](https://github.com/sakaki-/bcmrpi3-kernel-bis/pull/6)) I'll review and merge it.

Necktwi, 2019-05-23:
Please update the image with the latest kernel. I shall modify it to `btrfs` root.

sakaki-, 2019-05-24:
When the next release comes out it will have either the current, or a later release kernel included (so should have `btrfs` support). I'm in the early stages of working on this now, expected release is sometime near the end of June. Will keep this issue open as a reminder until then.
Best, sakaki

sakaki-, 2019-06-20:
The new (1.4.2) release is now out, and its kernel contains built-in support for `btrfs`.
Closing now. Please re-open if you still experience issues.
Thanks,
sakaki

closed 2019-06-20 by sakaki-