sakaki-/gentoo-on-rpi-64bit #88: RPI3B+ awfully slow after fresh 1.4.1 image install

zd59, 2019-05-22:
Hello!

Yesterday I installed your latest 1.4.1 image on fast (C10+) 16GB SD card and inserted in RPI3B+
Internet speed is 100/3 Mb/S - wired connection not by WiFi.
After waiting for a first run/setup (all left default) I added new user & reboot.
All went flawlessly without errors.
Waiting for a several minutes to settle down, I run Chrome - Web page open takes several minutes with a lots of mouse & keyboard freezes for several 10 seconds.
I tried root GUI package manager (portage) reload database (or something similar) and this almost freeze the system for almost two hours, with a very small number of few seconds time frames where the mouse/keyboard was responsive. 
Tried also to open Firefox alone (no other application except xfce running) - the same case as  Chrome. 
Conclusion:  My system as is not usable . Every action takes minutes. Upper right CPU load window did not show any traces of a load. Very strange.

How to find out what's failed?

sakaki-, 2019-05-22:
zd59,

sorry to hear you are having problems with this ><

First off, I would try using a vanilla copy of the image, _without_ a new user set up (i.e., just running as `demouser`), and see if you still get the same issue in e.g. Chromium. 

> The current version of the image (1.4.1) is in use by a relatively large user base (including myself!) and the sorts of lag you mention have not been reported elsewhere. Indeed others have reported the browser performance as _better_ than the stock system, see e.g. [this post](https://www.raspberrypi.org/forums/viewtopic.php?p=1467170#p1467170).

I would also try running a standard Raspbian OS on your RPi3 and browsing (using Chrome) to the same site, to check there are no hardware related issues.

Can you share the URL that you are having trouble with?

zd59, 2019-05-22:
Will try to run demouser when get back home (test in 5 hours).
HW is OK, with your older distribution it runs OK - posted questions here before.
XBian_2019.05.17_rpi3.img runs excellent, with no lag.

Tested URLs:
[(https://distrowatch.com/)](url)
[https://www.google.com](url)
[https://mail.google.com](url)

zd59, 2019-05-22:
Hi sakaki!

With Demo User there is no delay with applications running.
Where could be the reason?

I've got porthole failure at sync package database. See the picture. All settings are unchanged from installation and internet access is OK.

[image]

sakaki-, 2019-05-24:
The sync error you are seeing appears to be due to the (vanilla) Portage system attempting to update its keys. I have just tried it on a fresh copy of the image (1.4.1) here, and the keys update OK. Have you any restrictive settings on your subnet firewall or similar?

zd59, 2019-05-24:
> Have you any restrictive settings on your subnet firewall or similar?

Please be more specific. 
Only essential ports are open on my firewall to the wild: sh, http(s), ftp..
All other ports are closed. Until now I've never had any problems. Debian updates, none of the applications complained, Window$ works..

sakaki-, 2019-05-24:
Yes, actually looking more closely at your output, it is attempting to use `hkps`, not `hkp`. The latter uses port 11371 by default, which is frequently blocked, but the former uses port 443 by default, which is normally OK.
Can you try updating `portage` (as a oneshot) first, and see if that makes a difference?

sakaki-, 2019-06-20:
The new 1.4.2 image is now out, so hopefully this issue is now resolved. Please re-open if not.
Best, sakaki

closed 2019-06-20 by sakaki-