sakaki-/gentoo-on-rpi-64bit #89: Unable to download latest build

kodeschreiber, 2019-05-27:
Good Evening!

    I am trying to download a copy of the 1.4.1 build, but cannot seem to get it to fully download. I do apologize if this is more of a GitHub related issue, but I wanted to see what end it may be on. I know my connection can be a bit slow where I am, but event using the '--limit-rate' option in wget, the download will fail, typically around 11MB. I can try an older version and/or the source code to see if it makes a difference.

Thanks!

sakaki-, 2019-05-27:
Hi @kodeschreiber,

sorry to hear you are having trouble with this ><

I have noticed myself that GitHub downloads have been _very_ slow over the last 24 hours or so; hopefully this is only a temporary issue. Unfortunately it is beyond my control.

You can also try downloading the image via PINN (it is called "Gentoo64" there) - this uses a different (non-GitHub) server for the image. Instructions for downloading PINN may be found [here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574).

Best, sakaki

kodeschreiber, 2019-05-28:
Good Evening (well, now morning :) ),

I did some digging around on SourceForge for the PINN codes and am currently acquiring a copy of the image. I saw that the root and boot images are built into separate files (which may come in handy for later QEMU Virtual Machine testing). I'll give it a go later using the 'partition-setup.sh' script and let you know how it goes. Thanks for providing a mirror download!

Best Regards,
Kodeschreiber

kodeschreiber, 2019-05-29:
I have good news: after fumbling around with the PINN-based installation, I decided to give the GitHub link another shot. It was still acting strange, so I composed a simple script to help get the download completed:
`/bin/false; until [ $? -eq 0 ]; do wget --retry-connrefused --waitretry=1 --read-timeout=20 --timeout=15 -t 0 --limit-rate=500k -U firefox -c https://github.com/sakaki-/gentoo-on-rpi3-64bit/releases/download/v1.4.1/genpi64.img.xz; done
`
(it's not pretty, but it worked :) )

What's better is that I followed your guide for installing the image to the SD/booting and everything appears to be operating extremely well. I appreciate the help!

closed 2019-05-29 by kodeschreiber