sakaki-/gentoo-on-rpi-64bit #91: RPi4 Compatibility?

tdubz1, 2019-06-24:
Will your Gentoo OS be compatible with the new RPi 4? The new Pi has a much faster processor and you can buy a model which has 4GB RAM. Just wondering because I am thinking of purchasing one and think that these specs will make a great difference to the performance of the Pi and would be very cool to have a 64-bit OS on it. Cheers, tdubz1

sakaki-, 2019-06-29:
The current version isn't compatible with the Pi4 as it stands. However, I have just received one of these new Pi boards, and have started to play with it.

So, no promises, but we'll see what comes along as regards a bootable image ^-^

sakaki-, 2019-07-03:
Just FYI, I've been able to bring the [gentoo-on-rpi3-64bit](https://github.com/sakaki-/gentoo-on-rpi3-64bit) userland up on an RPi4 under a `bcm2711_defconfig` kernel (brief notes [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1491136#p1491136)).

Screenshot:
[image]

This is _only_ a proof of concept: any production-quality release will be gated on upstream progress on the 64-bit kernel, which (at this time) is not ready for prime time (1GiB memory clamp, SD card issues etc.) So we're realistically 1 to 2 months away from that.

Best, sakaki

tdubz1, 2019-07-04:
Sounds good! Looking forward to the future of this!

xlazom00, 2019-07-25:
It will be nice if we have whole image compiled specifically for cortex A72
and other one for cortex A53 to see difference.

sakaki-, 2019-07-25:
I am working on the 1.5.0 release of the image now; as you may have seen, the 64-bit kernel autobuilds for the Pi4 are [now in place](https://www.raspberrypi.org/forums/viewtopic.php?p=1506858#p1506858), which is a major element of this, so things are getting closer.

As to the userland binaries, I intend to switch from `-march=armv8-a+crc -mtune=cortex-a53` to `-march=armv8-a+crc -mtune=cortex-a72` on the 1.5.0 image, and for the binhost going forward. Since the `arch` is common (and only `mtune` is changing) between the Pi3 and Pi4 processors, code generated in this way _will_ still run on the older generation Pi3 board (per the `gcc` docs), but will be optimized for the out-of-order pipeline provided by the newer Pi4 SoC. I think this is a valid approach, since most people will shift to Pi4s in time, and the performance difference penalty (of running A72 tuned code on an A53) is likely to be small. And in any event,  I don't have the support bandwidth to maintain a full binhost for two separate mtune variants going forward.

There's also an argument for sticking with A53 tuning of course, since that's the slower processor where the performance difference will be more noticeable, but, having weighed things up, I have come to the conclusion (as just mentioned) that shifting to A72 tuning is the better approach.

sakaki-, 2019-08-07:
Brief update on this - the **full userland has now been rebuilt** under `-march=armv8-a+crc -mtune=cortex-a72`, and **pushed to the [binhost](https://www.isshoni.org/pi64pie/?C=M;O=D)**. The Portage rsync server has also been updated (new timestamp 4 Aug 2019).

> So, if you know what you are doing you can now try to install `tbz2` binary packages for the new `mtune` ^-^

With kernel autobuilds and the ground-up (`emptytree`) userland rebuild done, image assembly for 1.5.0 now in progress. Still some things to complete, like an updated [`pyconfig_gen`](https://github.com/sakaki-/pyconfig_gen) application etc. But, ETA around 2 weeks to release, barring real-world NMIs.

Best, sakaki

jdtsmith, 2019-08-16:
Does this build preserve OMX/MMAL capability?

sakaki-, 2019-08-17:
@jdtsmith, MMAL isn't something I've looked at recently - my bandwidth has been saturated getting the 1.5.0 image together tbh (now in hard freeze), and the v4l2{,m2m} stuff provides basic camera / h/w codec access for 64-bit userland right now.

The best thing might be to post a question on the raspberry pi forums, perhaps in [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=232684) (although it is a bit old now), to query the status. Last time I did look at this, there were user/kernel space pointer-size mismatch issues which prevented a 64-bit userland from accessing MMAL/OpenMAX IL correctly (all distros, see e.g. [this post](https://www.raspberrypi.org/forums/viewtopic.php?p=1433751#p1433751) by 6by9, _ff_), but tbh I'm not sure if these have recently been addressed. Sorry I can't be of more help on this one right now ><

Best, sakaki

sakaki-, 2019-08-19:
**1.5.0 is shipped**
Enjoy ^-^

Best, sakaki

sakaki-, 2019-08-20:
Closing now.

closed 2019-08-20 by sakaki-

sakaki-, 2019-10-25:
@jdtsmith,

an RPi-engineer-authored PR for 64-bit MMAL userland (with the pointer remapping etc.) is now available for test (raspberrypi/userland#586). So hopefully this feature will be available soon, certainly for release v1.5.2 of this image.

hth, sakaki

jdtsmith, 2019-10-25:
Thanks.  In the meantime @tmm1 has managed a workaround using the ISP hardware via V4L2.  But this could be interesting.