sakaki-/gentoo-on-rpi-64bit #93: Server image?

sedlund, 2019-07-16:
Could you make a stripped down 'server' image?

sakaki-, 2019-07-23:
Apologies for the delay in responding - I'm currently working on getting a pi3/pi4 image together.

Once that's out of the way, I will consider putting a "server" image together, as it is a request I've had a number of times.

In the meantime, there are instructions for stripping down the image yourself on the project's wiki [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit/wiki/Create-a-Slimmed-Down-Version-of-the-Image), which may be of use.

Best, sakaki

sakaki-, 2019-08-20:
OK I'm having a look at a stripped down, CLI-boot version now.

What do people generally think would be useful to have as dev tools on there? Just the baseline `@system` stuff, like `gcc`? Or the full `clang`, `icedtea`, `rust` etc?

Suggestions by Aug 27th please; I'll aim to lock off on something then.

Best, sakaki

sedlund, 2019-08-20:
I think as basic as possible.

sakaki-, 2019-08-20:
> I think as basic as possible.

@sedlund, so basically something similar to a stage 4 (but with the RPi-specific kernels, boot services etc. present), is that it?

Best, sakaki

sedlund, 2019-08-20:
Yes.  Would be nice to add the raspbian style systems service units to configure ssh and wifi as well.

chraaroe, 2019-08-21:
Agree with @sedlund. As basic as possible with the RPi kernels and services present.

chrbayer, 2019-08-23:
Yes, as basic as possible, just network, without wifi, config.txt optimised for CLI system with as much memory as possible. Probably the default kernel and not the bis kernel as default. OpenRC instead of systemd would be Ok for me.

chrbayer, 2019-08-23:
And maybe a 32-bit ARM 7l version, too... For my Rpi2 :-)

sakaki-, 2019-08-23:
OK that's all pretty clear, thanks.

So, I've split out a provisional `innercore` package set ([here](https://github.com/sakaki-/genpi64-overlay/blob/233d643bd8c8d63ac19fc6e9e4afac41013c8ddc/dev-embedded/rpi-64bit-meta/rpi-64bit-meta-1.5.1.ebuild#L69-L112)) from the old `core`; please let me know if there's anything missing / surplus to requirments on there; the minimal image will just have the transitive dependency closure of this plus the `@system` set.

Also, you really want the default rather than -bis kernel (actually kernel**s**, since there'll be a `kernel8.img` for the RPi3 and `kernel8-p4.img` for the RPi4, either way)?

Best, sakaki

ttyS0, 2019-08-23:
I think the `innercore` set looks good! 

My only question is about systemd. It there a reason beyond preference for it not being used? What I'm getting at is whether trying to convert this rpi4 setup from OpenRC to systemd is something that can be done by following the Gentoo systemd documentation, or whether there are known roadblocks that make it a fools errand.

chrbayer, 2019-08-24:
@sakaki- Don't you think the default kernels would be a good idea for a minimal image?

Do we really want to include the swap stuff in the innercore? For Rpi4 with 4 GB I would rather disable the swap for all images, guess an SD card it is slow and only damages the card with time...

sakaki-, 2019-08-26:
@chrbayer
the reason for the swap services is really to get `zswap` (which also needs some kernel config only in the `-bis` kernels). On memory constrained machines this can give a significant boost to the usable memory. Not so critical on 2GiB/4GiB Pi4Bs, where all swap may safely be disabled, but I've found it to be very useful - even for headless instances - on the RPi3.

hughjfchen, 2019-08-31:
@sakaki- 
May I ask if the "server" image is available for download? If yes, where is the url?

sakaki-, 2019-08-31:
@hughjfchen 

there will be a 'lite' (aka 'server') image available as part of the v1.5.1 release, which is in final test now, and should be out by the end of this weekend.

best, sakaki

hughjfchen, 2019-08-31:
@sakaki- 
Thanks for your update. Looking forward to the 1.5.1 release!

sakaki-, 2019-09-01:
**v1.5.1 now released**, with first cut at a 'lite'/'server' image available.
Feedback welcome!

sakaki-, 2019-09-02:
Following [comments](https://www.raspberrypi.org/forums/viewtopic.php?p=1529626#p1529626), I have modified the 'lite' image's partition sizing, so it will fit on microSD cards marked as >=8GB. The full image still requires a card marked as >=16GB.

The contents of the 'lite' image have not been changed (full package list [here](https://github.com/sakaki-/gentoo-on-rpi-64bit/blob/master/reference/world-all-packages-lite)).

Best, sakaki

chraaroe, 2019-09-04:
Hi @sakaki-. Great work!

I have a few questions regarding the packages for innercore. 

Why are stuff like alsa-tools required for the innercore USE flag? I also see stuff like xdm and xconsole which seems to be pulled in by rpi3-init-scripts. Are these strictly required?

sakaki-, 2019-09-04:
Probably not strictly required - I'll need to take a look at ensuring all the deps are minimized cross-sectionally for `innercore`. My goal here was just to get an initial iteration released so people could start using it (and feed back). I'll come back to this again later, once I have revisited other pending RPi4-related things (such as updating [raspbian-nspawn-64](https://github.com/sakaki-/raspbian-nspawn-64) for Buster etc.)

If you experiment, and have any suggestions for further thinning it down in the meantime, feel free to drop me a PR.

> Please note that I'm not going to be able to provide variant-USE flag builds for packages though, to minimize the support burden of this. Apologies, sakaki ><

hughjfchen, 2019-09-05:
@sakaki- 
Shall we add docker package into the "server" image package list? After all, that's a "server" image and intended to run some server applications.

yoshi314, 2019-09-05:
@hughjfchen  i think everyone has their own concept of a server. i'd rather have a minimal base to start from, you can always install docker from binpkg.

hughjfchen, 2019-09-05:
@sakaki- 
Fair enough. Thanks for your effort on this image. It helps me a lot getting start on my RPI4B.

sakaki-, 2019-11-10:
Closing now.

closed 2019-11-10 by sakaki-

samip5, 2019-12-12:
Random question, which relates to the server/lite image, is sshd enabled by default?

sakaki-, 2019-12-13:
Yes, it is.

Best, sakaki