sakaki-/gentoo-on-rpi-64bit #94: (not an error) Is the advice on placing a swap on another bus still relevant?

yrabbit, 2019-07-24:
Do I understand correctly that the subsystem responsible for allocating memory in the swap itself uses dynamically allocated memory and therefore all my attempts to put a swap on a USB flash fail under load?

[image]

sakaki-, 2019-08-17:
@yrabbit, apologies for not responding to you earlier on this, been busy on the forthcoming 1.5.0 release.

Interesting point! I must say I haven't seen that particular error come up, and I run a few build servers (which at times hit the swap pretty hard) using the dual-swap setup.

If you haven't already, why not post this question on the [RPi forums](https://www.raspberrypi.org/forums/)? I'd be interested myself to see what responses the RPF/RPT engineers come back with.

Best, sakaki

yrabbit, 2019-08-17:
I remembered what I did (compiled FreeCAD), I will try to repeat this in the near future in order to get the latest logs.

closed 2019-08-17 by yrabbit