sakaki-/gentoo-on-rpi-64bit #95: Rsync server out of date.

z-rui, 2019-08-01:
`emerge --sync` failed with the following message:
```
>>>                                                                                                                                                            
>>> SERVER OUT OF DATE: rsync://82.221.139.201/gentoo-portage-pi64-gem                                                                                         
>>>                                                                                                                                                            
>>> In order to force sync, remove '/usr/portage/metadata/timestamp.chk'.                                                                                      
>>>                                                                                                                                                            
```
The server-side timestamp seems to be Jun 13.

sakaki-, 2019-08-01:
Yes, apologies for that, it has been frozen for a while during the prep for the Pi 4 compatible 1.5.0 release. Should be unlocked (with updated contents) within a few weeks.
Best, sakaki

sakaki-, 2019-08-07:
So, I have just pushed a full update to the rsync server. The new timestamp is 4 Aug 2019.
You may get some issues trying to `genup` using this - it is a working version that will shortly be used for the basis of a new 1.5.0 release of the image, so quite possibly will need some additional profile tweaks etc. for correct build. All hopefully ironed out within the next 2 weeks.
Best, sakaki

sakaki-, 2019-08-07:
Closing now - please re-open if further issues.
Thanks!
sakaki

closed 2019-08-07 by sakaki-

nicolasvcr, 2020-09-25:
I just installed getoo-on-rpi today, stumbled against the same issue