sakaki-/gentoo-on-rpi-64bit #96: what about discord channel for this distro

Akiafroo, 2019-08-04:
i have couple of month using your gentoo with my pitop v2 and works great.! i remembered that you told in readme that if someone have one please let you know. well i have one and how i said works great.

my only issue is myself i try harder to learn about gentoo but something easy as install "audacity" gave a lot of headache i follow the howto but is impossible. for that i have the suggestion why not a discord channel where can follow more close this gentoo for pi3 and for the future of pi4 that i saw you are working on. ?

the discord channel its just a suggestion feel free to choose the best for you. thx. for all the work around this gentoo.

sakaki-, 2019-08-07:
Many thanks for letting me know!

Just to double check, does shutdown work correctly to power off the PiTop hub, and does the battery monitor plugin work correctly (top right of the screen)?

Best, sakaki

AkiAfroo, 2019-08-08:
the previous version turned off pitop v2 without problems in the latter does not turn off. i cant check it now because im doing some "Do it yourself" ("DIY") mods with the case. pi4 is out and the pitop company told that the pitop v2 will not get any upgrade like new hub etc,etc,. xD .

i will love to see an archlinux or debian derivate arm64 with all the stuff you put on. i tried 5 times gentoo and is not for me. xD thx in advance.

sakaki-, 2019-08-17:
I plan to put out an updated version of [raspbian-nspawn-64](https://github.com/sakaki-/raspbian-nspawn-64) after the 1.5.0 Gentoo image is released, that might be of some interest (the current version should work for you on a Pi-Top too; the host userland is Raspbian Stretch, but the guest is Debian Buster).

@satmandu also has a nice Docker-based Ubuntu aarch64 image builder for the Pi4, which may be of interest if you move to that board ([here](https://github.com/satmandu/docker-rpi4-imagebuilder)).

hth, sakaki

PS I'll close this issue now, please feel free to re-open if required. S.

closed 2019-08-17 by sakaki-