sakaki-/gentoo-on-rpi-64bit #97: Possible to show appreciations for your effort?

chraaroe, 2019-08-07:
Hi Sakaki, thank you for your effort on providing an awesome way of getting up and running with Gentoo on the RPi3. Also, it's thrilled to hear that you are planning to continue the work with the RPi4.

Is there any official way to support and/or thank you for your effort? An Amazon wishlist, a PayPal link or anything similar?

sakaki-, 2019-08-07:
@chraaroe,
many thanks for your kind comments ^-^
TBH positive feedback like this is thanks enough, no need for donations or similar.
But, I appreciate the thought!
Kind regards,
sakaki

chraaroe, 2019-08-08:
In that case, allow me to shower you with all the appreciation I can muster :)

If at some point you change your mind, please announce it. I would love to give a more tangible proof of appreciation.

Again, thank you for your effort.

closed 2019-08-08 by chraaroe